conorpp/u2f-zero #17: Consider crowdfunding

mkroman, 2016-06-23:
Hi - I love the project, thanks for making it! Already looking forward to building one soon.

Have you considered starting a crowdfunding campaign to get a low-cost u2f solution into the hands of the people?

I was thinking something like submitting it to http://www.seeedstudio.com/propagate/app/start/ and getting a quotation, then maybe create a project on https://www.crowdsupply.com/ to gather the funds and manufacture in larger bulk at a much lower cost.

With that said, I have no experience with manufacturing or crowdfunding - I just think it's something people would be interested in.

conorpp, 2016-06-23:
I'm currently in the process of ordering a small batch to be assembled after which I'll probably put on Amazon.

Crowd funding seems cool; I'm just not all convinced it would be necessary for this project.  If the order I'm currently in the process of eventually sells out then it will be worth it to look more into this.

I can create a timeline for this in the wiki if other people want to be involved.

jpouellet, 2016-06-24:
[int3.cc](http://int3.cc/) ([about](http://int3.cc/pages/about-us)) is relevant too, but I think you're on the right track already.

closed 2016-08-15 by conorpp