conorpp/u2f-zero #20: Stuck at "configuring..." when creating the u2f-firmware.hex

land0ngreer, 2016-07-05:
Not sure what I am missing. I could be missing a bunch but just following directions, unsure what in the world I am missing. Any help would be greatly appreciated! :)

_Getting Private key
dickbutt@dickbutmachine:~/Downloads/u2f-zero-master/tools/gencert/ca$ cd ../..
dickbutt@dickbutmachine:~/Downloads/u2f-zero-master/tools$ ./setup_device.sh gencert/ca/key.pem
configuring..._

conorpp, 2016-07-05:
The script stuck running this loop

```bash
while [[ "$?" -ne "0" ]] ; do
    sleep .2
    client.py configure pubkey.hex >/dev/null
done
```

Can you run `./u2f_zero_client/client.py configure pubkey.hex` and report what the output is?

Thanks

land0ngreer, 2016-07-05:
python hidapi module is required
try running: 
     apt-get install libusb-1.0-0-dev libudev-dev
     pip install hidapi

After installing the above, all is set. Thank you for the quick reply!

closed 2016-07-05 by land0ngreer