conorpp/u2f-zero #23: Add a way to make it updateable via USB

jumoog, 2016-07-12:
add FTDI FT2232 to make the u2f-zero updateable via USB (using the C2/JTAG interface of the EFM8UB1 series)

conorpp, 2016-07-12:
I'm not sure what you mean "using the C2/JTAG interface of the EFM8UB1 series".  It's can always be updated via the C2 interface.

To update over USB, there needs to be a bootloader.  There is probably a 1-2 KB of program memory leftover that the bootloader could be written in.  However, I don't think it would be any more convenient of a DIY project as the initial bootloader must be programmed via the C2 interface anyways.

jumoog, 2016-07-12:
i dont want spend money on a extra programmer

closed 2016-07-12 by jumoog

conorpp, 2016-07-13:
Assuming that everyone has an Arduino lying around, you could get this project to work:

https://github.com/x893/C2.Flash

Happy hacking!

jpouellet, 2016-07-15:
From a security perspective I do not like the idea of it being re-programmable over USB. The more complex of an interface you expose from the device to the machine you plug it into, the larger your effective attack surface.

The code required for the USB stack is non-trivial, and I would even be in favor of punting that to a separate USB->serial FTDI chip (or Cortex-M0 or something, because of FTDI's "anti-piracy" shenanigans), but the trade off there is increased cost per device.