conorpp/u2f-zero #24: settings for making project in Simplicity Studio

bensteinberg, 2016-07-25:
I'm having trouble building the project in Simplicity Studio. I got past some errors by installing the Keil license, but I'm still seeing some problems: one is `*** ERROR C249 IN LINE 368 OF /Users/ben/Documents/code/u2f-zero/firmware/src/u2f_atecc.c: 'DATA': SEGMENT TOO LARGE` -- when I change the memory model (in project properties, C/C++ build settings) from small or compact to large, I get `*** ERROR C202 IN LINE 57 OF /Users/ben/Documents/code/u2f-zero/firmware/lib/efm8ub1/bsp/efm8_memory_lcd/src/spi.c: 'BSP_DISP_CS': undefined identifier`. Can you tell me what your settings are for building this project?

conorpp, 2016-07-26:
You should not need to change the memory model.

I'm a little confused for your second error because `u2f-zero/firmware/lib/efm8ub1/bsp` does not exist in the repo and should not be there.  Compiling more code then necessary would cause an issue.

Double check that the optimization level is 9 and that optimization favors size (properties -> settings -> C/C++ build -> optimization).  Also the device should be shown as EFM8UB10F16G in project explorer.

tbh I would start over and reimport the project now that you have the Keil license worked out.

bensteinberg, 2016-07-26:
Thanks! I deleted, re-cloned, and imported the project -- now I'm getting the following five errors:

    *** ERROR L127: UNRESOLVED EXTERNAL SYMBOL
    SYMBOL:  __attest
    MODULE:  ./src/u2f_atecc.OBJ (U2F_ATECC)

    *** ERROR L127: UNRESOLVED EXTERNAL SYMBOL
    SYMBOL:  __attest_size
    MODULE:  ./src/u2f_atecc.OBJ (U2F_ATECC)

    *** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  __attest
    MODULE:  ./src/u2f_atecc.OBJ (U2F_ATECC)
    ADDRESS: 1003C86H

    *** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  __attest
    MODULE:  ./src/u2f_atecc.OBJ (U2F_ATECC)
    ADDRESS: 1003C88H

    *** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  __attest_size
    MODULE:  ./src/u2f_atecc.OBJ (U2F_ATECC)
    ADDRESS: 1003C50H

I have the device and the debug adapter connected, but I presume it's not necessary for the build step.

conorpp, 2016-07-26:
You need to program the setup build file first and then run configuration
script to generate the attestation certificate (__attest symbol).

See the steps in the wiki. Building a U2F token -> Programming.

bensteinberg, 2016-07-26:
Ah, of course. I did that, but didn't copy the certificate over from the first attempt. I'll try it this evening when I get back to that computer.

bensteinberg, 2016-07-27:
OK, I got it to compile -- I think I may have messed up the attestation certificate, though. I'll close this ticket, solder up another one, and start from scratch. Thanks for your help!

closed 2016-07-27 by bensteinberg

conorpp, 2016-07-27:
You can just reprogram the setup.hex file rerun the configuration to make
another attestation certificate and it will still work.

bensteinberg, 2016-07-29:
Thanks, working fine now -- it might even have been working fine earlier, except that the device never pulsed a green light, only red. I've checked the device at https://demo.yubico.com/u2f, though, so I know it works. I may have soldered the LED in backwards.... BTW, the photo at the top of https://github.com/conorpp/u2f-zero/wiki/Building-a-U2F-Token is of an old revision of the board, looks like. I'd offer up a picture of mine as a current example if I were more convinced of its correctness.

conorpp, 2016-07-29:
Nice.  Yes the pcb photo is on an out of date pcb but all the part
orientations are the same.  If the red dot on the LED is oriented on the
left side then it is correct.  Another soldering issue could be that the
red and green pins got bridged.

If you can get a good image of your board (which is the latest pcb) that
would be great to replace the old image on the wiki.

bensteinberg, 2016-08-03:
Thanks. I had the LED rotated 90 degrees. Rework on a hot surface did the trick. I've been trying to get a good photo, but haven't managed one yet with decent lighting, nice soldering, clean component placement, and visible component markings. I'll try some more, and have more PCBs coming from Dirty PCBS if I can't get a good shot of one of my first three devices.