conorpp/u2f-zero #25: [Hardware] Key Ring Durability

erichahn, 2016-09-08:
I've been having some issues with how much the keyring hole has been worn away.
[image]
[image]

martenwikman, 2016-09-10:
I think it would be a good idea if the board was extended a little so the hole would have more clearance to the board edge. Hole should be plated and some VIAs inserted in the pad itself like the picture below ( picture stolen from https://blogs.mentor.com/tom-hausherr/blog/tag/pcb-mounting-holes/) 
[image]

jaredfolkins, 2016-09-25:
Instead of modifying any design I used an aftermarket keychain solution for low friction. The wire is wrapped in plastic.

https://www.amazon.com/bayite-BYT-WKC-4INP12-Inches-Stainless-Keychains/dp/B01CLGTBFA

Works fine and the wear is not severe nor do I anticipate the device to fail or become lost because of the wear in the keyring hole. 

YMMV

alleybeachelectric, 2016-12-21:
Is there a 3d printable case that could be used to protect the PCB? If not, I don't mind creating one ;)

conorpp, 2016-12-25:
Yes, although it might not be very good.  Improvements/replacements welcome :)
https://github.com/conorpp/u2f-zero/tree/master/hardware/case

rspeed, 2017-01-16:
I wonder if something like Plasti Dip would be a good solution for this.

conorpp, 2017-01-25:
Updated PCB to use a plated hole / via as suggested by @martenwikman

closed 2017-01-25 by conorpp