conorpp/u2f-zero #28: [Question] Porting u2f-zero to Digispark

Akasurde, 2016-09-29:
This is rather a question than a issue.
Can we port this port this code to [DigiSpark](http://digistump.com/category/1) ? It would be nice.

ccoenen, 2016-09-29:
This circuit uses a [cryptographic coprocessor](http://www.atmel.com/Images/Atmel-8923S-CryptoAuth-ATECC508A-Datasheet-Summary.pdf) which is not on the digispark. At the very least this would have to be added. I'm not sure the ATtiny on the Digispark could even talk to it. It does not have I2C, so you'd have to [jump through some hoops](http://playground.arduino.cc/Code/USIi2c) to get this set up first.

I'm not directly involved in this project, so a more detailed answer would have to come from the actual author.

conorpp, 2016-09-29:
The code was written so it could be ported easily.

It does use an I2C peripheral to handle ECC and SHA256 operations.  So you would have to fill in the API calls like [u2f_ecdsa_sign](https://github.com/conorpp/u2f-zero/blob/master/firmware/inc/u2f.h#L162), [u2f_new_keypair](https://github.com/conorpp/u2f-zero/blob/master/firmware/inc/u2f.h#L170), [u2f_sha256_start](https://github.com/conorpp/u2f-zero/blob/master/firmware/inc/u2f.h#L170), etc. with your own implementation.

The final U2F Zero firmware is about 15 KB which is already more than the 8 KB of the ATtiny.  Adding software implementations of ECC operations and SHA256 would be impossible.  Even if you could, it would be very slow -- on the scale of 1-10 seconds per ECC operation.

closed 2016-09-29 by conorpp

Akasurde, 2016-10-03:
@ccoenen @conorpp Thanks for info.