conorpp/u2f-zero #30: Difficulties with HID USB

saeugetier, 2016-10-31:
I build two U2F by myself. PCBs were ordered from DirtyPCB in beginning of October. I think that the very latest gerbers are used, because I ordered the PCBs via link in the wiki.

I soldered all parts on the PCB. The soldering joints are checked with a microscope. I found no short circuits. 

After flashing SETUP.hex to board the USB connection is very unstable.

Sometimes the device is found, but most of the times the kernel is not able to initialize the USB device. dmesg | tail has following messages:

[72252.317662] usb 3-2: new low-speed USB device number 87 using xhci_hcd
[72252.318180] usb 3-2: Device not responding to setup address.
[72252.522419] usb 3-2: Device not responding to setup address.
[72252.726103] usb 3-2: device not accepting address 87, error -71
[72252.726177] usb usb3-port2: unable to enumerate USB device

I tried to compile the setup-firmware by myself. I got the same problem for both boards. 

The debugger identifies the microcontroller as "EFM8UB11F16G"

conorpp, 2016-10-31:
Seems odd, some ideas:

- Maybe the USB pins are not soldered well
- Maybe the USB exposed pads are not making strong contact with the USB socket in your computer.  Try another port or holding it in firmly.

If no success, can you post a picture?

saeugetier, 2016-10-31:
I have tested the sticks on two different computers. For tests I plugged the stick in every single slot.

I have tested the resistance of the signal D+ and D-. The resistance between connector pad and "wire" on the micro controller housing is zero ohm.

Do I have to program any fuses? For example for the clock sources?

conorpp, 2016-10-31:
I just read you only flashed SETUP.hex -- did you finish the rest of the programming steps?

saeugetier, 2016-10-31:
[image]

saeugetier, 2016-10-31:
The SETUP.hex has to be flashed first. After flashing the stick should enumerate as HID adapter.

conorpp, 2016-10-31:
Hm, does the LED glow green when powered?

saeugetier, 2016-10-31:
The LED glows green. While pressing the button it is blue.

conorpp, 2016-11-13:
I have some DirtyPCBs on the way for the different project, so after I assemble them I can check if I can replicate the issue.  Will get back.

bmkillian, 2016-11-13:
Sounds good.  Just an FYI – this is a unit I purchased on Amazon, not one I
built myself. Thanks!

*From:* Conor Patrick [mailto:notifications@github.com]
*Sent:* Saturday, November 12, 2016 8:18 PM
*To:* conorpp/u2f-zero
*Subject:* Re: [conorpp/u2f-zero] Difficulties with HID USB (#30)

I have some DirtyPCBs on the way for the different project, so after I
assemble them I can check if I can replicate the issue. Will get back.

—
<https://github.com/conorpp/u2f-zero/issues/30#issuecomment-260162069>, or mute
the thread

conorpp, 2016-11-13:
@bmkillian this is for a different issue

bmkillian, 2016-11-13:
I’m an idiot.  My apologies.

*From:* Conor Patrick [mailto:notifications@github.com]
*Sent:* Saturday, November 12, 2016 8:23 PM
*To:* conorpp/u2f-zero
*Cc:* Brandon; Mention
*Subject:* Re: [conorpp/u2f-zero] Difficulties with HID USB (#30)

@bmkillian <https://github.com/bmkillian> this is for a different issue

—
<https://github.com/conorpp/u2f-zero/issues/30#issuecomment-260162245>, or mute
the thread

conorpp, 2018-01-04:
Sorry I never got around to this and I haven't heard about any similar issues.  There's been a layout update since this issue which likely helps with USB signal integrity

closed 2018-01-04 by conorpp