conorpp/u2f-zero #31: Unable to build firmware

raoulh, 2016-10-31:
Trying to build the firmware with simplicity studio 4.0 but it failed.
After doing file->import the u2fzero firmware into the workspace, the toolchain/board/part are not available. It seems the project does not have a correct config somehow.

How to build it?

conorpp, 2016-10-31:
Did you install the EFM8 device when installing simplicity studio?

saeugetier, 2016-10-31:
Have you tried Simplicity Studio 3 or 4? I had issues with version 4. Please try version 3.

raoulh, 2016-10-31:
Yes. I also can create a new project and build it correctly.

I found out that the project files:
https://github.com/conorpp/u2f-zero/blob/master/firmware/.project#L23
does contain this instead in my working project:
`<nature>com.silabs.ss.framework.ide.project.sls.core.SLSProjectNature</nature>`

No idea why... Do you build in the latest simplicity studio 4.0.1?

raoulh, 2016-10-31:
Version 4 here.

raoulh, 2016-10-31:
Ok, works with Simplicity Studio 3

ThomasWaldmann, 2016-11-06:
So, if it doesn't work with v4, add the version number to the u2f-zero docs?

conorpp, 2016-11-13:
I tried recreating the project in v4 but there's another problem with the program not compiling to a small enough size.  Program size grew by about 2KB from v3 to v4, which is quite significant for a 16KB memory microcontroller.   Silabs probably added a lot more code to the v4 SDK.

So only v3 will work.  I've update wiki.

closed 2016-11-13 by conorpp

ThomasWaldmann, 2016-11-13:
Thanks for updating the docs!