conorpp/u2f-zero #32: Memory error w/ pamu2fcfg

yangwao, 2016-11-01:
Any idea what's happening?
Can I print more verbose output?
```
wao@astrid~> pamu2fcfg -d
USB send: 00ffffffff8600080807060504030201000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
USB read rc read 64
USB recv: ffffffff8600110807060504030201cafebac2020200000300000000000000000000000000000000000000000000000000000000000000000000000000000000
device USB_10c4_8acf_14200000 discovered as 'U2F Zero'
  version (Interface, Major, Minor, Build): 2, 2, 0, 0  capFlags: 3
JSON: { "challenge": "gY6upYqKqqSSZdIBimxsZntlK_3rslYBPjE3l2QBL60", "version": "U2F_V2", "appId": "pam:\/\/astrid.local" }
JSON challenge URL-B64: gY6upYqKqqSSZdIBimxsZntlK_3rslYBPjE3l2QBL60
client data: { "challenge": "gY6upYqKqqSSZdIBimxsZntlK_3rslYBPjE3l2QBL60", "origin": "pam:\/\/astrid.local", "typ": "navigator.id.finishEnrollment" }
JSON: { "challenge": "gY6upYqKqqSSZdIBimxsZntlK_3rslYBPjE3l2QBL60", "version": "U2F_V2", "appId": "pam:\/\/astrid.local" }
JSON app_id pam://astrid.local
USB send: 00cafebac28300470001030000004027872bd2fea557304d0e5666b286215ac58fd28f8ff11576e0f4222777aa8f3e88b509d12a3183116802af3e1f00039438
USB write returned 65
USB send: 00cafebac200a2f5249547ec997c43e48a6f9d200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
now trying with timeout 8
now trying with timeout 16
now trying with timeout 32
now trying with timeout 64
now trying with timeout 128
now trying with timeout 256
now trying with timeout 512
USB read rc read 64
USB recv: cafebac2830002921000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB data (len 2): 9210
error: (-1) Memory error (e.g., out of memory)
```

conorpp, 2016-11-01:
Hi Matej,

I'm glad that the shipment made it.  I believe that is because the key is
registered to 15 services and is out of memory.  It only has enough WO
memory on secure chip for 15 private keys.  You can reset the registrations
by using the Python script client.  See "extras" in wiki.

Conor

yangwao, 2016-11-01:
Oh, yes! Thanks! :)
Well, I have struggle with that python utils, but I think it's b/c of my system (Darwin Kernel Version 16.0.0), but on linux it worked :)
```wao@astrid~/P/u/t/u2f_zero_client> ./client.py wipe
python hidapi module is required
try running:
     apt-get install libusb-1.0-0-dev libudev-dev
     pip install hidapi

yangwao, 2016-11-01:
I just as instructed, but this just failed for me 🎱  hints & tips?
```
wao@astrid~/P/u/t/u2f_zero_client> ./client.py wipe
opened  None
Press U2F button until the LED is no longer red.
Traceback (most recent call last):
  File "./client.py", line 279, in <module>
    do_wipe(h)
  File "./client.py", line 234, in do_wipe
    res = h.read(64, 10000)
  File "hid.pyx", line 122, in hid.device.read (hid.c:2686)
IOError: read error
```

conorpp, 2016-11-01:
Sometimes when you press the button it can momentarily disconnect power which breaks the wipe -- does it always do that?

yangwao, 2016-11-01:
Everytime, I just hold my thumb on button, it just blinks from bright red into bright blue then back to bright red and suddenly to bright blue and then dimmed blue. Sometimes there is missing second red blink and just went from first red into bright blue.
Oh and I just figured it out, I just need press it more times.
Maybe it should be there _Press **repeatedly** U2F button_
```
wao@astrid~/P/u/t/u2f_zero_client> ./client.py wipe
opened  None
Press U2F button until the LED is no longer red.
Wipe succeeded
```

closed 2016-11-02 by conorpp