conorpp/u2f-zero #35: SimplicityStudio - not that simple

ThomasWaldmann, 2016-11-06:
At least not using Ubuntu 16.04. Their setup wants to install conflicting packages.
```
# note: twinst.sh is about the same as the apt-get install in their setup.sh.
# I tried without the mesa / gl stuff because that triggered similary issues in my previous try.
# But even without these it breaks due to conflicts in the zlib / libz32 stuff.
tw@tux:~/SimplicityStudio_v3$ sudo ./twinst.sh 
[sudo] password for tw: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
lib32z1 is already the newest version (1:1.2.8.dfsg-2ubuntu4).
libsdl1.2debian is already the newest version (1.2.15+dfsg1-3).
libasound2:i386 is already the newest version (1.1.0-0ubuntu1).
libbz2-1.0:i386 is already the newest version (1.0.6-8).
libbz2-1.0:i386 set to manually installed.
libxtst6:i386 is already the newest version (2:1.2.2-1).
libopenal1 is already the newest version (1:1.16.0-3).
libopenal1 set to manually installed.
libsdl-image1.2 is already the newest version (1.2.12-5build2).
libsdl-image1.2 set to manually installed.
libc6-i386 is already the newest version (2.23-0ubuntu4).
libc6-i386 set to manually installed.
libuuid1:i386 is already the newest version (2.27.1-6ubuntu3.1).
libuuid1:i386 set to manually installed.
The following additional packages will be installed:
  gcc-5-base:i386 lib32tinfo5 libatk1.0-0:i386 libaudio2:i386
  libavahi-client3:i386 libavahi-common-data:i386 libavahi-common3:i386
  libboost-filesystem1.58.0:i386 libboost-system1.58.0:i386 libbsd0:i386
  libcairo2:i386 libcups2:i386 libdatrie1:i386 libdrm-amdgpu1:i386
  libdrm-dev:i386 libdrm-intel1:i386 libdrm-nouveau2:i386 libdrm-radeon1:i386
  libdrm2:i386 libedit2:i386 libegl1-mesa:i386 libegl1-mesa-dev:i386
  libevdev2:i386 libgbm1:i386 libgdk-pixbuf2.0-0:i386 libgl1-mesa-dev:i386
  libgl1-mesa-dri:i386 libgl1-mesa-glx:i386 libglapi-mesa:i386
  libgles2-mesa:i386 libgles2-mesa-dev:i386 libglib2.0-0:i386
  libglu1-mesa:i386 libglu1-mesa-dev:i386 libgmp10:i386 libgnutls30:i386
  libgraphite2-3:i386 libgssapi-krb5-2:i386 libgudev-1.0-0:i386
  libharfbuzz0b:i386 libhogweed4:i386 libicu55:i386 libidn11:i386
  libinput10:i386 libjbig0:i386 libjpeg-turbo8:i386 libjpeg8:i386
  libk5crypto3:i386 libkeyutils1:i386 libkrb5-3:i386 libkrb5support0:i386
  liblcms2-2:i386 libllvm3.8:i386 libmirclient-dev:i386 libmirclient9:i386
  libmircommon-dev:i386 libmircommon5:i386 libmircookie-dev:i386
  libmircookie2:i386 libmirprotobuf3:i386 libmng2:i386 libmtdev1:i386
  libmysqlclient20:i386 libnettle6:i386 libpango-1.0-0:i386
  libpangocairo-1.0-0:i386 libpangoft2-1.0-0:i386 libpciaccess0:i386
  libpcre16-3:i386 libpixman-1-0:i386 libprotobuf-dev:i386
  libprotobuf-lite9v5:i386 libprotobuf9v5:i386 libproxy1v5:i386
  libpthread-stubs0-dev:i386 libqt4-dbus:i386 libqt4-declarative:i386
  libqt4-network:i386 libqt4-script:i386 libqt4-sql:i386 libqt4-sql-mysql:i386
  libqt4-xml:i386 libqt4-xmlpatterns:i386 libqt5concurrent5:i386
  libqt5core5a:i386 libqt5dbus5:i386 libqt5gui5:i386 libqt5network5:i386
  libqt5opengl5:i386 libqt5opengl5-dev:i386 libqt5printsupport5:i386
  libqt5sql5:i386 libqt5sql5-sqlite:i386 libqt5svg5:i386 libqt5test5:i386
  libqt5widgets5:i386 libqt5xml5:i386 libqtcore4:i386 libqtdbus4:i386
  libsqlite3-0:i386 libthai0:i386 libtiff5:i386 libtxc-dxtn-s2tc0:i386
  libudev1:i386 libwacom-bin libwacom2:i386 libwayland-client0:i386
  libwayland-cursor0:i386 libwayland-dev:i386 libwayland-egl1-mesa:i386
  libwayland-server0:i386 libx11-dev:i386 libx11-xcb-dev:i386 libx11-xcb1:i386
  libxau-dev:i386 libxcb-dri2-0:i386 libxcb-dri2-0-dev:i386 libxcb-dri3-0:i386
  libxcb-dri3-dev:i386 libxcb-glx0:i386 libxcb-glx0-dev:i386
  libxcb-icccm4:i386 libxcb-image0:i386 libxcb-keysyms1:i386
  libxcb-present-dev:i386 libxcb-present0:i386 libxcb-randr0:i386
  libxcb-randr0-dev:i386 libxcb-render-util0:i386 libxcb-render0:i386
  libxcb-render0-dev:i386 libxcb-shape0:i386 libxcb-shape0-dev:i386
  libxcb-shm0:i386 libxcb-sync-dev:i386 libxcb-sync1:i386 libxcb-util1:i386
  libxcb-xfixes0:i386 libxcb-xfixes0-dev:i386 libxcb-xkb1:i386
  libxcb1-dev:i386 libxcomposite1:i386 libxcursor1:i386 libxdamage-dev:i386
  libxdmcp-dev:i386 libxext-dev:i386 libxfixes-dev:i386 libxi6:i386
  libxkbcommon-dev:i386 libxkbcommon-x11-0:i386 libxkbcommon0:i386
  libxshmfence-dev:i386 libxshmfence1:i386 libxxf86vm-dev:i386
  libxxf86vm1:i386 mesa-common-dev:i386 mysql-common qt-at-spi:i386
[... 681 lines cut ...]
Selecting previously unselected package libqt5opengl5-dev:i386.
Preparing to unpack .../libqt5opengl5-dev_5.5.1+dfsg-16ubuntu7.2_i386.deb ...
Unpacking libqt5opengl5-dev:i386 (5.5.1+dfsg-16ubuntu7.2) ...
Selecting previously unselected package libqt5sql5-sqlite:i386.
Preparing to unpack .../libqt5sql5-sqlite_5.5.1+dfsg-16ubuntu7.2_i386.deb ...
Unpacking libqt5sql5-sqlite:i386 (5.5.1+dfsg-16ubuntu7.2) ...
Selecting previously unselected package libwacom-bin.
Preparing to unpack .../libwacom-bin_0.18-1_amd64.deb ...
Unpacking libwacom-bin (0.18-1) ...
Selecting previously unselected package qt-at-spi:i386.
Preparing to unpack .../qt-at-spi_0.4.0-3_i386.deb ...
Unpacking qt-at-spi:i386 (0.4.0-3) ...
Processing triggers for libc-bin (2.23-0ubuntu4) ...
Processing triggers for libglib2.0-0:amd64 (2.48.1-1~ubuntu16.04.1) ...
Processing triggers for man-db (2.7.5-1) ...
Errors were encountered while processing:
 /var/cache/apt/archives/zlib1g-dev_1%3a1.2.8.dfsg-2ubuntu4_i386.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

In a previous try, i tried to "fix" the conflict by explicitely installing zlib1g-dev:386 and didn't notice that it wanted to remove a ton of stuff - that almost destroyed my system and it tool me an hour to manually repair the damage again.

Of course this is not u2f-zero's fault, but I thought I give feedback in case other users run into same issues.

My first try was with SimplicityStudio v4 and after I read in the issues tracker that v3 works better, I had quite similar issues with v3 (see log above).

ThomasWaldmann, 2016-11-06:
Shorter log showing the issue (done after the above steps):
```
$ sudo apt-get install -f
[sudo] password for tw: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  zlib1g-dev:i386
The following NEW packages will be installed:
  zlib1g-dev:i386
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
173 not fully installed or removed.
Need to get 0 B/167 kB of archives.
After this operation, 403 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
(Reading database ... 332533 files and directories currently installed.)
Preparing to unpack .../zlib1g-dev_1%3a1.2.8.dfsg-2ubuntu4_i386.deb ...
Unpacking zlib1g-dev:i386 (1:1.2.8.dfsg-2ubuntu4) ...
dpkg: error processing archive /var/cache/apt/archives/zlib1g-dev_1%3a1.2.8.dfsg-2ubuntu4_i386.deb (--unpack):
 trying to overwrite '/usr/include/i386-linux-gnu/zconf.h', which is also in package lib32z1-dev 1:1.2.8.dfsg-2ubuntu4
Processing triggers for man-db (2.7.5-1) ...
Errors were encountered while processing:
 /var/cache/apt/archives/zlib1g-dev_1%3a1.2.8.dfsg-2ubuntu4_i386.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

ThomasWaldmann, 2016-11-06:
Known bug in Ubuntu: https://bugs.launchpad.net/ubuntu/+source/zlib/+bug/1512992

Suggested workaround there:
```
sudo apt-get -o Dpkg::Options::="--force-overwrite" install zlib1g-dev:i386
```

conorpp, 2016-11-13:
Thanks, I added this to a trouble shooting page in wiki

https://github.com/conorpp/u2f-zero/wiki/Trouble-shooting

closed 2016-11-13 by conorpp