conorpp/u2f-zero #40: Doesn't register on Windows through FIDOboosterUno

scateu, 2016-12-01:
http://www.vancosys.com/fidobooster/
                                              
When I try to register it through "FIDOboosterUno for Windows 10 64bit"，U2F-Zero does give an indication for me to press the button. But FIDOboosterUno doesn't continue after I pressed it. 
(The U2F-Zero device still keep the indication lights on) 

While it works well for my Yubikey Neo.  

Plus, what can I do if I want to debug this issue?

Thanks.

darconeous, 2016-12-01:
Have you confirmed that your token works for other sites, like Google? I wonder if the button on yours might be broke.

scateu, 2016-12-02:
It works for google, dropbox,fastmail, even for pam.d(linux/macOS), and firefox u2f plugin.

So it can be sure there is some compatibility issue with FIDObooster.

I have written to them, and get a response that they are working on it.

crrodriguez, 2016-12-05:
@scateu maybe you are hitting https://github.com/conorpp/u2f-zero/issues/36 ? can you check if the key is full already ?

scateu, 2016-12-06:
Actually, I tried after 'python client.py wipe' (and 5 press)...

0. `python client.py wipe`
1. Register a new key via FIDO Booster
2. U2F Zero: LED, Red
3. me pressing U2F Zero button, LED turns Blue
4. me releasing U2F Zero button, LED turns back to Red
5. FIDO Booster stills prompt "please press the button on your key"
6. ...... after pressing button for 15 times, LED indicates that key is full

conorpp, 2016-12-06:
@scateu to wipe you need to press button 5 consecutive times.  The client tool will let you know if it worked.

scateu, 2016-12-06:
"Actually, I tried after 'python client.py wipe' (and 5 press)..."

and it did wiped.

conorpp, 2016-12-06:
I just checked the debugging output and registration seems fine.  It's likely an issue with FIDO Booster not handling something right.  For instance, when all 15 slots are use up, the u2f zero replies with a valid error response indicating there is no memory.  Yet FIDO booster continues to send repeated registration requests.

The program probably has the typical try/except while loop.  It probably runs into an error parsing the u2f zero registration certificate which is different from Yubico's cert, causing it to resend another registration.  Yubico is probably what the FIDO Boost authors tested against.

scateu, 2016-12-06:
Thanks. I have reported this issue to Vancosys. Hope this information will 
help them debug FIDO booster.

(CC: Bahram and Geoff from vancosys.com)

conorpp, 2017-01-28:
Is this still being looked at or can I close?

scateu, 2017-02-04:
I am trying to program the latest firmware into U2F Zero. After that, I'll test it again.

conorpp, 2018-01-04:
Assuming I can close this, let me know if there are still issues.

closed 2018-01-04 by conorpp

GitHubMad, 2018-03-22:
On Windows 10, I also didn't succeed using U2F-zero with **FIDOboosterUno**.
I bought my U2F-zero [here](https://www.amazon.fr/U2F-Zero-U2-F/dp/B01L9DUPK6) around Xmas.
Tested in february.
So maybe issue should be re-open...

conorpp, 2018-03-22:
There was a faulty batch shipped around Christmas that was fixed by early January.  Email me your order ID and we can go from there