conorpp/u2f-zero #41: DirtyPCBs links dead

kurtcarpenter, 2016-12-08:
Looks like the folks at DirtyPCBs redid their site, now neither of the links on the wiki Parts List work anymore. Can the wiki be updated with the new links? (found by searching the Shares page)

Panelized: http://dirtypcbs.com/store/designer/details/9836/997/u2f-zero-panel-gerbers-zip
Non-panelized: http://dirtypcbs.com/store/designer/details/9836/998/u2f-zero-gerbers-zip

conorpp, 2016-12-08:
Updated!  Thanks!

DurandA, 2017-09-28:
Out of curiosity, what PCB thickness did you get (using 2mm option) from *dirtypcbs*? I got some 1.85mm PCBs and these are very loose in the USB socket.

rickard-von-essen, 2017-09-29:
Some weeks ago I got a batch of 2 mm PCB's from DirtyPCB which where 2.06mm, the fits nicely. (And as a reference my Yubikey NEO is 2.33mm which is probably close to optimal.)

conorpp, 2017-09-29:
DirtyPCBs uses a number of different board houses which is probably the reason for the inconsistency.  Reopening issue to try to find a better PCB source to get something around 2.1-2.2 mm

Suggestions welcome

DurandA, 2017-10-02:
I'll try [FirstPCB](https://firstpcb.com/) and report back. Unfortunately only 2mm is available and I cannot find other suppliers with 2.1-2.3mm thickness without prohibitive prices.

What is the thickness of the unit sold on Amazon? Unfortunately I cannot order it in my country.

conorpp, 2017-10-02:
It's about 2.05 mm.  2.1-2.2 would probably be better.

DurandA, 2017-10-08:
I have very good experience with [printed.cz](http://www.printed.cz/eng/shop/set-full-mode). I just noticed they offer 2.2mm for an affordable price. However, they don't have low cost slow shipping options so it may be too expensive for hobbyists wanting to build a few units.

athavan-mahen, 2017-11-23:
Hi where can I get a downloadable version of these files  
u2f-zero-panel-gerbers-zip, 
u2f-zero-gerbers-zip 
so I can try to use a different vendor for printing or to make some changed 

Thank you