conorpp/u2f-zero #45: Is the programmer are really required?

ad-m, 2017-01-22:
Hello,

[Silicon Labs EFM8UB11F16G-C-QSOP24 datasheet](http://www.mouser.com/ds/2/368/EFM8UB1_DataSheet-793165.pdf) mention:
> All devices come pre-programmed with a USB bootloader. This bootloader resides in the code security page and last pages of code flash; it can be erased if it is not needed.

Is there a way to program a microcontroller through USB without development board or debugger mentioned in [Parts List](https://github.com/conorpp/u2f-zero/wiki/Parts-List)? 

Greetings,
Greetings,

ad-m, 2017-01-22:
[AN945: EFM8 Factory Bootloader User Guide](https://www.silabs.com/Support%20Documents/TechnicalDocs/AN945.pdf) mention:
> All devices come pre-programmed with a UART (~500 bytes), SMBus (~500 bytes), or **USB HID (~1.5 kB) bootloader**. This bootloader resides in the code security page. If it's too large to fit in the code security page, it will also occupy the last pages of code flash. It can be erased if it is not needed.

conorpp, 2017-01-22:
Hi,

The Silabs provided bootloader seems to start in the security page (64 bytes) then jump to regular program memory.  Just about all of the program memory is used by the application so there's not really room for both.  Maybe with some careful tinkering, it could be possible to fit both (USB HID code reuse between bootloader and application, etc).  Another option could be to use the bootloader to load the application to configure the crypto chip and then overwrite everything when the final application is loaded.  In this case, you can only program the final application once without the need of extra hardware.

So yes with some more work at least _something_ in this direction could be worked out.

conorpp, 2017-01-28:
Just added support to use USB HID bootloader.  [Extra functionality](https://github.com/conorpp/u2f-zero/wiki/Extras-and-Usage) has to be remove for it all to fit.  See [wiki](https://github.com/conorpp/u2f-zero/wiki/Building-a-U2F-Token) for instructions.

closed 2017-01-28 by conorpp