conorpp/u2f-zero #46: Register u2f-0 without phone?

richrumble, 2017-01-24:
Are there any services that allow you just use the u2f, without first using a phone? That is the use-case I bought this for, but as far as I can tell, Google/Github require you use a phone to enable 2fa, then you can use your u2f. Is there a way to test to see if it's working? I tried the fidobooster from vasco, it didn't see either of my devices when I pressed the buttons. I don't need/use dropbox etc, I wanted this for Google and Git exclusively. Windows seems to have the driver, chrome is ver 55...
-rich

richrumble, 2017-01-24:
Will u2f-0 work on the Yubico test site?
https://demo.yubico.com/start/u2f/neo
Mine don't seem to if they should work.

AdrianKoshka, 2017-01-24:
The only thing I could think of right off would be the `PAM` stuff for linux (I use it on my main desktop for 2FA login w/ lightdm, though I don't use a U2F-0 key specifically), otherwise I can't think of services that don't require a phone to enable 2FA then setup U2F.

richrumble, 2017-01-24:
Thanks! I was able to use a different PC and the Yubico/demo app as a test, they both appear to be working from that perspective. There is something on that other PC that is blocking the u2f-0. AFAIK a phone will have to be used first in order to use u2f-0 on these two services.
This was the link that actually worked: https://demo.yubico.com/u2f

closed 2017-01-24 by richrumble