conorpp/u2f-zero #48: ./setup_device.sh failed.

scateu, 2017-02-04:
With the latest commit. Using programmer.

After I erase U2F Zero with Simplicity Studio, I program U2F Zero with `setup.hex`

Then:

```
$ ./setup_device.sh gencert/ca/key.pem gencert/ca/cert.der 
configuring...
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 102, 69, 0, 0, 80, 0, 213, 168, 139, 196, 238, 192, 85, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
conf:  0123664500005000d5a88bc4eec05500c0005500837181018371c101837183718371c171010183718371c1718371837183718371ffffffff00000000ffffffff00000000fffffffffffffffffffffffffff
fffff00005555ffff00000000000013003c0013003c0013003c0013003c003c003c0013003c0013003c0013003300
crc is  ['0xc9', '0x3b']
locked eeprom with crc  [201, 59]
failed writing master key
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 102, 69, 0, 0, 80, 0, 213, 168, 139, 196, 238, 192, 85, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
conf:  0123664500005000d5a88bc4eec05500c0005500837181018371c101837183718371c171010183718371c1718371837183718371ffffffff00000000ffffffff00000000fffffffffffffffffffffffffff
fffff00005555ffff00000000000013003c0013003c0013003c0013003c003c003c0013003c0013003c0013003300
crc is  ['0xc9', '0x3b']
locked eeprom with crc  [201, 59]
^CTraceback (most recent call last):
  File "./u2f_zero_client/client.py", line 344, in <module>
    do_configure(h, sys.argv[2],sys.argv[3])
  File "./u2f_zero_client/client.py", line 203, in do_configure
    time.sleep(0.250)
KeyboardInterrupt
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 102, 69, 0, 0, 80, 0, 213, 168, 139, 196, 238, 192, 85, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

```
$ uname -a
Linux ThinkPad 4.4.0-57-lowlatency #78-Ubuntu SMP PREEMPT Sat Dec 10 01:37:35 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

conorpp, 2017-02-08:
Thanks, there is a bug in setup.  I will look into fixing soon.

conorpp, 2017-02-10:
I forgot to update the fixed setup.hex files.  It seems to work now.

scateu, 2017-02-10:
Hi, 

I just tried with the new commit.

```
$ ./setup_device.sh gencert/ca/key.pem  gencert/ca/cert.der 
configuring...
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 102, 69, 0, 0, 80, 0, 213, 168, 139, 196, 238, 192, 85, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
conf:  0123664500005000d5a88bc4eec05500c0005500837181018371c101837183718371c171010183718371c1718371837183718371ffffffff00000000ffffffff00000000ffffffffffffffffffffffffffffffff00005555ffff00000000000013003c0013003c0013003c0013003c003c003c0013003c0013003c0013003300
crc is  ['0xc9', '0x3b']
locked eeprom with crc  [201, 59]
failed writing master key
```

While in serial port:

```
U2F ZERO
0123664500005000d5a88bc4eec05500c000550083a083a083a083a083a083a0
U2F ZERO
U2F ZERO
0123664500005000d5a88bc4eec05500c000550083a083a083a083a083a083a0
83a083a083a083a083a083a083a083a083a083a000030000000a0009ffffffff
00000000ffffffffffffffffffffffffffffffff00000000ffff000000000000
1300130013001300130013001300130013001300130013001300130013003300
config crc:ed67 
00000000
already locked
81000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
U2F_CONFIG_IS_BUILD
80000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
U2F_CONFIG_GET_SERIAL_NUM
83c93b00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
got crc: c93b 
00000000
config already locked
00000000
data already locked
855de572f0e5b2d4ac7f8ceb6ca7dad684c606acbb9fe92192e25608467f15f5db00000000000000000000000000000000000000000000000000000000000000
U2F_CONFIG_LOAD_TRANS_KEY
master key: 5de572f0e5b2d4ac7f8ceb6ca7dad684c606acbb9fe92192e25608467f15f5db
writing master key failed
error: f 
```

Actually, I tried to build `setup.hex` from scratch, and the result didn't change.

conorpp, 2017-02-10:
Old U2F zeros will not work with the new firmware since I changed the configuration of the ATECC508A.  It can't be changed after it has been locked.  So you can do 1 of two things:

1.  Desolder the ATECC508A and replace with a new one.  Then reprogram.
2.  Email me your address and I can ship you a new one.

scateu, 2017-02-10:
OK.
I will replace the old ATECC508A. 

Thanks again for your great job!

closed 2017-02-10 by scateu