conorpp/u2f-zero #50: IndexError: list index out of range

dl8bb, 2017-02-10:
Hi folks,
  looks like I run into a similar issue than described in #48 . 

```
./setup_device.sh gencert/ca/key.pem gencert/ca/cert.der
+ SETUP_HEX=../firmware/SETUP.hex
+ FINAL_HEX=../firmware/release/u2f-firmware.hex
+ FLASH_TOOLS=0
+ SN=
+ SN_build=
+ SN_setup=
+ [[ 2 != \2 ]]
+ attest_priv=gencert/ca/key.pem
+ attest_pub=gencert/ca/cert.der
+ [[ 2 != \2 ]]
+ export PATH=/home/bergt/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:gencert:u2f_zero_client:flashing
+ PATH=/home/bergt/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:gencert:u2f_zero_client:flashing
+ [[ 0 = 1 ]]
+ echo configuring...
configuring...
+ [[ -n '' ]]
+ client.py configure gencert/ca/key.pem pubkey.hex
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 146, 137, 0, 0, 80, 0, 6, 60, 106, 15, 238, 192, 101, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
conf:  0123928900005000063c6a0feec06500c0005500837181018371c101837183718371c171010183718371c1718371837183718371ffffffff00000000ffffffff00000000ffffffffffffffffffffffffffffffff00005555ffff00000000000013003c0013003c0013003c0013003c003c003c0013003c0013003c0013003300
crc is  ['0xad', '0x9']
data is  []
Traceback (most recent call last):
  File "u2f_zero_client/client.py", line 345, in <module>
    do_configure(h, sys.argv[2],sys.argv[3])
  File "u2f_zero_client/client.py", line 199, in do_configure
    if data[1] == 1:
IndexError: list index out of range
+ [[ 1 -ne 0 ]]
+ sleep .2
+ [[ -n '' ]]
+ client.py configure gencert/ca/key.pem pubkey.hex
```

For me it looks like the read_n_tries() is returning no data.

BTW: I added -x flag to the shell bang to get more info about what's going on and also a print statment in u2f_zero_client/client.py so the line numbers are increased from original.

...
    196     data = read_n_tries(h,5,64,1000)
    197     print( 'data is ', [hex(x) for x in data])
    198
    199     if data[1] == 1:
    200         print( 'locked eeprom with crc ',crc)
...

Do you have any hints on how to fix this?

Kind regards,
birko
--

conorpp, 2017-02-10:
Hey Birko,

If can add an option for better verbosity and make a pull request, that would be great.

Are you using an old U2F Zero?

> Old U2F zeros will not work with the new firmware since I changed the configuration of the ATECC508A.  It can't be changed after it has been locked.  So you can do 1 of two things:
> 
> 1.  Desolder the ATECC508A and replace with a new one.  Then reprogram.
> 2.  Email me your address and I can ship you a new one.

dl8bb, 2017-02-10:
Hi,
  what do you mean by a old U2F zero? I ordered the PCB from dirtypcbs.com at the beginning of Jan. 2017 and got them this week. I already build some u2f-zero sticks and try to work through the init procedure. 

BTW: I used the SETUP.hex from the github to flash it. Just to try if I can flash. Was this the mistake? ...and I bricked my ATECC508A? :-(

regards,
birko
--

conorpp, 2017-02-10:
If you used firmware with commit [fe9cc54](https://github.com/conorpp/u2f-zero/tree/fe9cc54a9784d5024db310a06b84b7a5a5e8a8e0) or before (January 25) then it would have written the old ATECC508A configuration and will not work with the newer firmware.  You can replace the ATECC508A to use latest firmware or you can revert back to commit fe9cc54.

I just added two releases to make this easier for people that need to distinguish.

closed 2017-04-29 by conorpp