conorpp/u2f-zero #53: SMD Parts are too damn small!

makefu, 2017-03-24:
[image]

I really like the project but i had much difficulty with soldering the parts onto the PCB. For one the parts you chose were just too damn small for soldering by hand. On the other hand the PCB itself is "huge" compared to the the parts placed. I'd love to see a version with somewhat bigger parts ( 0805 ), especially the five pins of the zener diode, the LED and the 0402 capacitor produced a lot of headache.

Besides that, keep up your work to make the community more secure!

jbheren, 2017-03-24:
I agree with that, especially the 0.1 capacitor & Diode, even if it can mak us proud when it works :) 
Soldering by hand may work fine with 1206. I made a test on pcb with a milling machine and some wires are too thin for it.

[image]

conorpp, 2017-03-24:
@makefu Yeah I agree.  I can look into adding larger components later but it would be best if someone else could do this.  I think the zener, 0402 cap, and LED would all be easy to replace with slightly bigger versions.

@jbheren I recently accepted a [pull request](https://github.com/conorpp/u2f-zero/pull/37) that normalized the wire widths (among other things) and should improve milling/fab yield.  You might have better luck with the latest version.

jbheren, 2017-03-26:
Thanks @conorp, I made a milling simulation an it sounds much better with this version. last time I tested using the panel file which seems quite older.

conorpp, 2018-05-04:
Fixed by #80 thanks to @thenoviceoof

closed 2018-05-04 by conorpp