conorpp/u2f-zero #54: USB HID Flashing / bootloader

jbheren, 2017-03-26:
Hi All, 

I made an U2F Zero token from one [OSHPARK ](https://oshpark.com/shared_projects/Rq9evtm7) made PCB & spent some time trying to flash it :

- flashed the SETUP.boot after I converted the SETUP.hex from repository,
- Generated certificate and ran the Configuration. I had some errors but the flow said " device configured" so I tried the next step,
- Set the device to bootloader mode (took some time to understand we had to set it using `client.py bootloader`  after keeping finger on the switch (red LED),
- Flashed the built u2f-firmare after commenting out the needed defines (as defined in wiki) : 

```
C:\Users\jb\Documents\AN945SW\Tools\Windows>efm8load.exe u2f-firmware.boot
Download over port: HID:10C4:EAC9

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@AAAAAAC@@

Download complete with [ 7 ] errors
```

I Think the token moved a bit as I had left my hand on the switch...

Now It seems like I am unable to set the device back to bootloader mode. Do you think I bricked it ?

Thanks again for all the big work.

conorpp, 2017-03-26:
If you tried running 

```bash
./u2f_zero_client/client.py bootloader
```

And it doesn't work, it may be bricked.  You can try programming the bootloader again with the efm8load.exe tool.  If it lost power during the flashing then it probably got corrupted and is why it won't go back into bootloader mode.

jbheren, 2017-03-26:
Ok thanks, then I will try to build another one. Do you think it could be recovered using a dedicated programmer ?

conorpp, 2017-03-26:
Yes you will have no issue with bricking it/recovering when using a programmer

closed 2017-03-27 by jbheren

Oliv4945, 2018-04-23:
Hi @jbheren, did succeed to program another chip ?
I have the exactly the same error, might be related to #57

Also, for your information, as written in #57, you can put the device in bootloader again by shorting C2D pin to ground when powering the key.

j8048188, 2018-07-10:
I've been having the same programming issues as @jbheren. I can reset the u2f by shorting C2D and Ground when i plug the key into my computer. I'm happy to provide any files requested to help debug this.