conorpp/u2f-zero #56: can't concat bytes to str in client.py:188

ad-m, 2017-04-02:
Hello,

I am tried to initialize some devices by USB programming. On Cywgin & Windows with Simplicity Studio 3 and .

As mentioned in the docs I executed:
``
cd ca/
./genca.sh
cd ../..
./setup_device.sh gencert/ca/key.pem gencert/ca/cert.der   # or path to your preferred attestation key pair
``

Then the LED is pulsing a green light and command line provides infinity output of errors like:
```
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 95, 16, 0, 0, 80, 0, 166, 147, 83, 58, 238, 192, 137, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Traceback (most recent call last):
  File "./u2f_zero_client/client.py", line 344, in <module>
    do_configure(h, sys.argv[2],sys.argv[3])
  File "./u2f_zero_client/client.py", line 188, in do_configure
    config = array.array('B',data[2:2+l]).tostring() + config[l:]
TypeError: can't concat bytes to str
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 95, 16, 0, 0, 80, 0, 166, 147, 83, 58, 238, 192, 137, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Traceback (most recent call last):
  File "./u2f_zero_client/client.py", line 344, in <module>
    do_configure(h, sys.argv[2],sys.argv[3])
  File "./u2f_zero_client/client.py", line 188, in do_configure
    config = array.array('B',data[2:2+l]).tostring() + config[l:]
TypeError: can't concat bytes to str
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 95, 16, 0, 0, 80, 0, 166, 147, 83, 58, 238, 192, 137, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Traceback (most recent call last):
  File "./u2f_zero_client/client.py", line 344, in <module>
    do_configure(h, sys.argv[2],sys.argv[3])
  File "./u2f_zero_client/client.py", line 188, in do_configure
    config = array.array('B',data[2:2+l]).tostring() + config[l:]
TypeError: can't concat bytes to str
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 95, 16, 0, 0, 80, 0, 166, 147, 83, 58, 238, 192, 137, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Traceback (most recent call last):
  File "./u2f_zero_client/client.py", line 344, in <module>
    do_configure(h, sys.argv[2],sys.argv[3])
  File "./u2f_zero_client/client.py", line 188, in do_configure
    config = array.array('B',data[2:2+l]).tostring() + config[l:]
TypeError: can't concat bytes to str
```
It references to [this line](https://github.com/conorpp/u2f-zero/blob/8931b3205b2b601a312053f92dd337a2877296d3/tools/u2f_zero_client/client.py#L188). 

Now on Linux there is no errors like that if I try reproduce the issue without U2F device. It might be Python / Cygwin incompatibility. 

```
In [1]: import array

In [2]: data = [128, 15, 1, 35, 95, 16, 0, 0, 80, 0, 166, 147, 83, 58, 238, 192, 137, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

In [3]: 

In [3]: l = data[1]

In [4]: array.array('B',data[2:2+l]).tostring()
Out[4]: '\x01#_\x10\x00\x00P\x00\xa6\x93S:\xee\xc0\x89'

In [5]: type(array.array('B',data[2:2+l]).tostring())
Out[5]: str

In [6]: type(config[l:])
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-6-dfbf22984aa3> in <module>()
----> 1 type(config[l:])

NameError: name 'config' is not defined

In [7]: config = "\x01\x23\x6d\x10\x00\x00\x50\x00\xd7\x2c\xa5\x71\xee\xc0\x85\x00\xc0\x00\x55\x00\x83\x71\x81\x01\x83\x71\xC1\x01\x83\x71\x83\x71\x83\x71\xC1\x71\x01\x01\x83\x71\x83\x71\xC1\x71\x83\x71\x83\x71\x83\x71\x83\x71\xff\xff\xff\xff\x00\x00\x00\x00\xff\xff\xff\xff\x00\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x55\x55\xff\xff\x00\x00\x00\x00\x00\x00\x13\x00\x3C\x00\x13\x00\x3C\x00\x13\x00\x3C\x00\x13\x00\x3C\x00\x3c\x00\x3C\x00\x13\x00\x3C\x00\x13\x00\x3C\x00\x13\x00\x33\x00"

In [8]: type(config[l:])
Out[8]: str

In [9]: array.array('B',data[2:2+l]).tostring() + config[l:]
Out[9]: '\x01#_\x10\x00\x00P\x00\xa6\x93S:\xee\xc0\x89\x00\xc0\x00U\x00\x83q\x81\x01\x83q\xc1\x01\x83q\x83q\x83q\xc1q\x01\x01\x83q\x83q\xc1q\x83q\x83q\x83q\x83q\xff\xff\xff\xff\x00\x00\x00\x00\xff\xff\xff\xff\x00\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00UU\xff\xff\x00\x00\x00\x00\x00\x00\x13\x00<\x00\x13\x00<\x00\x13\x00<\x00\x13\x00<\x00<\x00<\x00\x13\x00<\x00\x13\x00<\x00\x13\x003\x00'
``
I have to dig a bit in week 3-9 April 2017 over that.

conorpp, 2017-04-03:
Interesting.  Maybe changing the line to something like this could be a possible fix?

```python
config = array.array('B',data[2:2+l]).tostring() + array.array('B',config[l:]).tostring()
```

conorpp, 2018-01-04:
I'm assuming this was fixed, let me know if there are still issues.

closed 2018-01-04 by conorpp