conorpp/u2f-zero #57: USB Problem after writting final compiled firmware.

racerxdl, 2017-04-13:
I followed the steps in the wiki and I'm getting this error when compiling:
```
*** WARNING L15: MULTIPLE CALL TO FUNCTION
    NAME:    _USB_WRITEFIFO/EFM8_USBDEP
    CALLER1: ?C_C51STARTUP
    CALLER2: USBIRQHANDLER/EFM8_USBDINT

*** WARNING L15: MULTIPLE CALL TO FUNCTION
    NAME:    _USBD_WRITE/EFM8_USBD
    CALLER1: ?C_C51STARTUP
    CALLER2: USBIRQHANDLER/EFM8_USBDINT
```

that leads to unrecognizable device:
```
[30067.125863] usb 1-5.2: new full-speed USB device number 112 using xhci_hcd
[30067.213843] usb 1-5.2: device descriptor read/64, error -71
[30067.401887] usb 1-5.2: device descriptor read/64, error -71
[30067.589863] usb 1-5.2: new full-speed USB device number 113 using xhci_hcd
[30067.673905] usb 1-5.2: device descriptor read/64, error -71
[30067.861880] usb 1-5.2: device descriptor read/64, error -71
[30068.053901] usb 1-5.2: new full-speed USB device number 114 using xhci_hcd
[30068.074414] usb 1-5.2: device descriptor read/8, error 0
[30068.194386] usb 1-5.2: device descriptor read/8, error 0
[30068.381877] usb 1-5.2: new full-speed USB device number 115 using xhci_hcd
[30068.398392] usb 1-5.2: device descriptor read/8, error 0
[30068.518371] usb 1-5.2: device descriptor read/8, error 0
[30068.622050] usb 1-5-port2: unable to enumerate USB device\
```

I disabled the 4 features in the header as mention (to not corrupt the bootloader) and if I compile with **ATECC_SETUP_DEVICE** defined (so setup mode) the generated hex file works fine (the USB device works, and u2f client list devices perfectly). If I disable **ATECC_SETUP_DEVICE** I get the compilation warning and the device doesn't work.

I'm using a programmer to reset everything when it goes bad, since I cannot reset it in the normal way.

racerxdl, 2017-04-13:
Just notice if I just leave the genca with default parameters (without adding anything) the key inside the firmware program is about 3 bytes less and it works. If I change any of the parameters on genca, it grows some bytes and stops working,

racerxdl, 2017-05-03:
Ok so its not size related, because now i rebuild the CA, gave it 18 more bytes (instead of 3) and its working. So it looks like some field cannot be filled on CA.

conorpp, 2017-05-03:
Thanks for sharing this.  The warning you get during compilation happen every time and shouldn't be an issue.  Are you still programming using the bootloader?

racerxdl, 2017-05-03:
Yeah, but when it doesn't work, it breaks the USB bootloader (since it get 3 bytes error writting the firmware), then I use the programmer to reset the device back to the bootloader. I will check some more parameters on `makeca` to see what causes the failure.

rickard-von-essen, 2017-09-23:
I think I get the same problem. Did you figure out what's causing this?

racerxdl, 2017-09-23:
Seens related to CA Cerfificate. I managed to change just the company name and got it working. Looks like its one of `makeca` parameters but I couldn't figured out which one.

conorpp, 2017-09-23:
There might just too much data with all the fields filled in the cert, causing the memory to overwrite the bootloader page.  Could you post what firmware (and cert) sizes worked and didn't work?

samop, 2018-02-10:
The problem still persists and I cannot program firmware into the controller. However, according to Application note AN945, bootloader can be invoked with by connecting C2D pin to ground. There is no need for programmer to fix the bootloader if something goes wrong during flashing.

Oliv4945, 2018-04-19:
Hi, I am running into the same issue. Did you find any solution ?
What are working values for `makeca` ? I am unable to generate valid parameters.

The U2F is correctly is recognized with setup.hex:
```
./client.py list
interface_number : -1
manufacturer_string : Silicon Labs
path : \\?\hid#vid_10c4&pid_8acf#6&2fbc42&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
product_id : 35535
product_string : U2F Zero
release_number : 256
serial_number : CAFEBABEFFFFFFFF
usage : 1
usage_page : 61904
vendor_id : 4292
```
I can do the setup:
```
 ./setup_device.sh gencert/ca/key.pem gencert/ca/cert.der                     
configuring...
opened  None
Device is configured.
read 15 bytes
[128, 15, 1, 35, 32, 168, 0, 0, 80, 0, 219, 100, 183, 198, 238, 192, 73, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
conf:  012320a800005000db64b7c6eec04900c0005500837181018371c101837183718371c171010183718371c1718371837183718371ffffffff00000000ffffffff00000000ffffffffffffffffffffffffffffffff00005555ffff00000000000013003c0013003c0013003c0013003c003c003c0013003c0013003c0013003300
crc is  ['0xf1', '0xa5']
locked eeprom with crc  [241, 165]
wkey e68a01ce0d14d40c7d2e299780f764e8ecdde0188de7cbb09bb6a8da8e33a273a6a678eb
writing keys to  pubkey.hex
Done.  Putting device in bootloader mode.
generate attestation certificate...
for file gencert/ca/cert.der
done.
building...
Open Simplicity Studio and rebuild final program.
Then you can erase and reprogram U2F Token.
```
I have the same warning during the link:
```
LX51 LINKER/LOCATER V4.66.30.0 - SN: K1RIC-S0KLQZ
COPYRIGHT ARM Germany GmbH 1995 - 2014
@u2f-firmware.lnp "./src/InitDevice.OBJ",
"./src/Interrupts.OBJ",
"./src/SILABS_STARTUP.OBJ",
"./src/bsp.OBJ",
"./src/callback.OBJ",
"./src/descriptors.OBJ",
"./src/u2f.OBJ",
"./src/u2f_hid.OBJ",
"./lib/efm8ub1/peripheralDrivers/src/usb_0.OBJ",
"./lib/efm8_usb/src/efm8_usbd.OBJ",
"./lib/efm8_usb/src/efm8_usbdch9.OBJ",
"./lib/efm8_usb/src/efm8_usbdep.OBJ",
"./lib/efm8_usb/src/efm8_usbdint.OBJ",
"./lib/efm8_assert/assert.OBJ",
"./tests/tests.OBJ",
"./src/atecc508a.OBJ",
"./src/cert.OBJ",
"./src/custom.OBJ",
"./src/eeprom.OBJ",
"./src/i2c.OBJ",
"./src/main.OBJ",
"./src/u2f_atecc.OBJ"
TO "U2F-FIRMWARE.OMF.CRBUILD" REMOVEUNUSED PRINT(.\u2f-firmware.m51) PAGEWIDTH (120) PAGELENGTH (65) CLASSES( CODE(C:0x0 - C:0x3ffe), CONST(C:0x0 - C:0x3ffe), ECODE(C:0x0 - C:0x3ffe), HCONST(C:0x0 - C:0x3ffe), XDATA(X:0x0 - X:0x3ff), HDATA(X:0x0 - X:0x3ff))

*** WARNING L15: MULTIPLE CALL TO FUNCTION
    NAME:    _USB_WRITEFIFO/EFM8_USBDEP
    CALLER1: ?C_C51STARTUP
    CALLER2: USBIRQHANDLER/EFM8_USBDINT

*** WARNING L15: MULTIPLE CALL TO FUNCTION
    NAME:    _USBD_WRITE/EFM8_USBD
    CALLER1: ?C_C51STARTUP
    CALLER2: USBIRQHANDLER/EFM8_USBDINT

Program Size: data=104.7 xdata=1013 const=578 code=14946
LX51 RUN COMPLETE.  2 WARNING(S),  0 ERROR(S)
Finished building target: u2f-firmware.omf
```

Then flashing the software reports errors. I do not find any documentation on how to get the error.
```
./efm8load.exe u2f-firmware.boot
Download over port: HID:10C4:EAC9

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@AAAAAAC@@

Download complete with [ 7 ] errors
```

Thank you
Oliv'

j8048188, 2018-07-10:
If you run efm8load.exe -t firmware.boot it'll give you the full download details, showing the errors.
My errors look liek the following: 
`C:\Users\Jared\Downloads\u2f\release>efm8load.exe -t firmware.boot
Download over port: HID:10C4:EAC9

$ 04 31 A5 F1 00 -> @
$ 83 32 00 00 FF 24 3B 7E -> @
$ 83 33 00 80 F6 08 DF FA -> @
$ 83 33 01 00 A3 C8 C5 82 -> @
$ 83 33 01 80 22 BB FE 02 -> @
$ 83 32 02 00 E0 38 F0 22 -> @
$ 83 33 02 80 82 C8 F2 E5 -> @
$ 83 33 03 00 02 03 4F 50 -> @
$ 83 33 03 80 A3 C5 83 C5 -> @
$ 83 32 04 00 F8 02 08 5A -> @
$ 83 33 04 80 82 88 83 E4 -> @
$ 83 33 05 00 F0 02 DF F4 -> @
$ 83 33 05 80 F5 80 3C 89 -> @
$ 83 32 06 00 83 CC E4 93 -> @
$ 83 33 06 80 68 E7 09 F2 -> @
$ 83 33 07 00 80 87 80 87 -> @
$ 83 33 07 80 E9 98 EA 9C -> @
$ 83 32 08 00 82 AB 83 23 -> @
$ 83 33 08 80 01 E5 F0 A3 -> @
$ 83 33 09 00 02 12 15 0D -> @
$ 83 33 09 80 FD 02 0A 0D -> @
$ 83 32 0A 00 0C 7F 01 22 -> @
$ 83 33 0A 80 14 90 EC 4D -> @
$ 83 33 0B 00 70 12 14 C7 -> @
$ 83 33 0B 80 74 F0 90 01 -> @
$ 83 32 0C 00 12 21 E2 12 -> @
$ 83 33 0C 80 E4 FD 80 05 -> @
$ 83 33 0D 00 C4 13 54 03 -> @
$ 83 33 0D 80 3C 79 9E ED -> @
$ 83 32 0E 00 70 37 EF 24 -> @
$ 83 33 0E 80 45 13 F5 C2 -> @
$ 83 33 0F 00 0B FD 80 52 -> @
$ 83 33 0F 80 26 57 90 03 -> @
$ 83 32 10 00 53 C0 03 C0 -> @
$ 83 33 10 80 B8 12 2F 67 -> @
$ 83 33 11 00 8D 90 02 6E -> @
$ 83 33 11 80 7E 69 7F 84 -> @
$ 83 32 12 00 12 03 B8 90 -> @
$ 83 33 12 80 73 24 C1 60 -> @
$ 83 33 13 00 24 74 6D 75 -> @
$ 83 33 13 80 7C 00 FD 02 -> @
$ 83 32 14 00 01 74 54 7F -> @
$ 83 33 14 80 DC E0 75 F0 -> @
$ 83 33 15 00 F5 83 22 90 -> @
$ 83 33 15 80 4E 90 02 6B -> @
$ 83 32 16 00 35 7A 02 79 -> @
$ 83 33 16 80 80 00 90 02 -> @
$ 83 33 17 00 8F F0 12 01 -> @
$ 83 33 17 80 AD 03 90 02 -> @
$ 83 32 18 00 00 79 21 22 -> @
$ 83 33 18 80 03 02 19 46 -> @
$ 83 33 19 00 E0 60 05 EF -> @
$ 83 33 19 80 00 74 E0 12 -> @
$ 83 32 1A 00 12 06 3A 90 -> @
$ 83 33 1A 80 02 CA E0 44 -> @
$ 83 33 1B 00 90 02 D8 E5 -> @
$ 83 33 1B 80 D0 E0 14 60 -> @
$ 83 32 1C 00 30 24 02 70 -> @
$ 83 33 1C 80 90 02 CE E0 -> @
$ 83 33 1D 00 60 06 12 3A -> @
$ 83 33 1D 80 90 03 B7 E0 -> @
$ 83 32 1E 00 7F 01 90 02 -> @
$ 83 33 1E 80 7B 01 7A 02 -> @
$ 83 33 1F 00 C7 80 3A 12 -> @
$ 83 33 1F 80 07 E4 90 02 -> @
$ 83 32 20 00 EA F0 A3 EB -> @
$ 83 33 20 80 70 0A 90 02 -> @
$ 83 33 21 00 40 22 D3 EF -> @
$ 83 33 21 80 F0 E4 FC FD -> @
$ 83 32 22 00 E0 64 06 60 -> @
$ 83 33 22 80 DD 12 03 B8 -> @
$ 83 33 23 00 2B F6 90 01 -> @
$ 83 33 23 80 FB 12 2D 6E -> @
$ 83 32 24 00 AF 10 E5 AE -> @
$ 83 33 24 80 40 80 90 29 -> @
$ 83 33 25 00 80 FB 30 06 -> @
$ 83 33 25 80 80 03 12 35 -> @
$ 83 32 26 00 60 07 E4 FB -> @
$ 83 33 26 80 33 78 02 C3 -> @
$ 83 33 27 00 7F 04 7A 02 -> @
$ 83 33 27 80 D0 03 D0 02 -> @
$ 83 32 28 00 1C 12 28 3F -> @
$ 83 33 28 80 90 03 B3 E0 -> @
$ 83 33 29 00 12 33 E9 80 -> @
$ 83 33 29 80 89 30 05 1D -> @
$ 83 32 2A 00 EA 41 02 E2 -> @
$ 83 33 2A 80 7F FF 22 E4 -> @
$ 83 33 2B 00 94 04 50 59 -> @
$ 83 33 2B 80 03 B8 74 FF -> @
$ 83 32 2C 00 90 01 DD 02 -> @
$ 83 33 2C 80 CF E0 FB 12 -> @
$ 83 33 2D 00 02 9F 12 03 -> @
$ 83 33 2D 80 F5 83 EE F0 -> @
$ 83 32 2E 00 EF F0 AA 06 -> @
$ 83 33 2E 80 04 08 7B FF -> @
$ 83 33 2F 00 06 3A 78 3D -> @
$ 83 33 2F 80 7E 00 7F 02 -> @
$ 83 32 30 00 A3 EF F0 C2 -> @
$ 83 33 30 80 60 3A A2 AF -> @
$ 83 33 31 00 F5 A7 22 90 -> @
$ 83 33 31 80 E4 F0 90 01 -> @
$ 83 32 32 00 E5 12 39 C1 -> @
$ 83 33 32 80 02 7F 47 12 -> @
$ 83 33 33 00 12 32 0E 02 -> @
$ 83 33 33 80 B9 A3 E0 FE -> @
$ 83 32 34 00 AF 20 E3 F3 -> @
$ 83 33 34 80 D0 E0 32 EF -> @
$ 83 33 35 00 03 4F 7F 00 -> @
$ 83 33 35 80 02 C8 E0 F5 -> @
$ 83 32 36 00 D3 22 8D 82 -> @
$ 83 33 36 80 30 03 0A 90 -> @
$ 83 33 37 00 80 02 7E 20 -> @
$ 83 33 37 80 75 A7 20 43 -> @
$ 83 32 38 00 1F EE 60 0B -> @
$ 83 33 38 80 37 40 7B FF -> @
$ 83 33 39 00 22 75 AE 91 -> @
$ 83 33 39 80 AF AF 22 75 -> @
$ 83 32 3A 00 75 A7 20 75 -> A
$ 83 33 3A 80 02 30 5B 31 -> A
$ 83 33 3B 00 30 5B 31 0B -> A
$ 83 33 3B 80 A7 03 AC C6 -> A
$ 83 32 3C 00 11 5C A4 74 -> A
$ 83 33 3C 80 00 40 C4 10 -> A
$ 13 33 3D 00 D5 FF 3C E9 -> A
$ 07 34 00 00 3D 0F CA C0 -> C
$ 04 33 00 00 02 -> @
$ 03 36 00 00 -> @

Download complete with [ 8 ] errors`