conorpp/u2f-zero #59: Adding an user-defined access key-press code to firmware?

BCLaraby, 2017-09-29:
This is a great project however the one concern I have is the actual physical loss of the key -- especially in the face of losing it in a public place or some sort of potentially malicious means. 

I don't know how it might work but perhaps there would be a way to allow, in firmware, for some sort of a user-designated key-press combination that enables the key once it's first plugged in, before it can be used?

rickard-von-essen, 2017-09-29:
U2F was specifically designed to be a second factor. Losing one factor shouldn't be fatal.

conorpp, 2017-10-07:
U2F could potentially be used as a single factor, in which case this could be a nice enhancement for some