conorpp/u2f-zero #64: Unable to add second U2F Zero to Google account.

ryanpcmcquen, 2017-11-13:
I thought this was an issue with Google's U2F implementation, but after reporting it on their help forum, I have heard reports of others adding multiple keys of the same type to the same Google account.

Relevant link:
https://support.google.com/chromebook/forum/AAAAmKCdEusTYUH4OFdOnQ/?hl=en

conorpp, 2017-11-14:
There was an issue with U2F Zero not working correctly when multiple U2F tokens are registered to one account.  The issue has recently [been fixed](https://github.com/conorpp/u2f-zero/commit/255c520e016caee914cfa878514cb60a126e6e0b) but all stock on Amazon is currently affected.

If you purchased one on Amazon, send me an email and I can send you a new one.

ryanpcmcquen, 2017-11-14:
Thank you @conorpp. Do you have a PGP key?

conorpp, 2017-11-15:
@ryanpcmcquen  yes: https://pgp.mit.edu/pks/lookup?op=vindex&search=0x1442045251A28169

ryanpcmcquen, 2017-11-27:
@conorpp, will you close this when the Amazon stock is updated?

conorpp, 2017-11-27:
What is currently on Amazon is up-to-date/fixed but I'll leave the issue open for now in case others have the same issue.

ryanpcmcquen, 2017-11-29:
@conorpp, just to clarify, if I order one today from Amazon it will not have this issue?

conorpp, 2017-11-30:
@ryanpcmcquen That's correct, at least from the U.S. market.

conorpp, 2018-01-04:
@ibotty Can you register it at https://demo.yubico.com/u2f and open up the technical information and post it here?

ibotty, 2018-01-04:
I'll do so in a second, but just for reference: How can I get that information from the shell?

ibotty, 2018-01-04:
```
Registration Data
origin: https://demo.yubico.com
version: U2F_V2
challenge: kJh_sWagZzvVWBuszvkTiAVPeLs0G1M1VQGM8YdQsUY
appId: https://demo.yubico.com

Response Data
clientData: {"challenge":"kJh_sWagZzvVWBuszvkTiAVPeLs0G1M1VQGM8YdQsUY","origin":"https://demo.yubico.com","typ":"navigator.id.finishEnrollment"}
registrationData: 0504204222dac0fa1dbc4104fefe520f94ba7e40aae0d39df297c7014fe51ad8ecae656f9dd3645bcb5209259d9e686fa6e7555eed2fd6928841c0060566c4d9ab1c24778452bdc1c842790d16504fa90742cb048f45dabd8a23cc38ebafe16e816f7fe9ffc25e308201de30820185020100300a06082a8648ce3d040302307b310b3009060355040613025553310b300906035504080c0256413113301106035504070c0a426c61636b73627572673110300e060355040a0c07436f6e6f72436f3114301206035504030c0b636f6e6f72636f2e636f6d3122302006092a864886f70d0109011613636f6e6f72636f40636f6e6f72636f2e636f6d301e170d3137303230323232303432345a170d3137303330343232303432345a307c310b3009060355040613025553310b300906035504080c0256413113301106035504070c0a426c61636b73627572673111300f060355040a0c08553246205a65726f3114301206035504030c0b7532667a65726f2e636f6d3122302006092a864886f70d0109011613636f6e6f72636f40636f6e6f72636f2e636f6d3059301306072a8648ce3d020106082a8648ce3d03010703420004a1155e6372c345ec97c12ac2ba84f863f1ae6ff17d800402a19f8606e422a7c02be4e34461153efbd41324ec84b254ec5852910d55b19b3079f588dbf305839a300a06082a8648ce3d04030203470030440220205c9e1d0d57cc2f0b25f0971cbf398ea3b20f9c9771fbfe88f4f9a103134cb6022026eedaaed8bf7c93700fcd1a264c52fae386293350b17a21112c03398dfde256304402203a640e21aba2ce8313056c2a3c28c23e682185cf8f16b5980fd9fb8a318a835202207fc2f204052a8fb7e5061d93369725f40d80b63b7829d315c1c99a9e877bc92d

Attestation Certificate
Certificate:
    Data:
        Version: 1 (0x0)
        Serial Number: 0 (0x0)
    Signature Algorithm: ecdsa-with-SHA256
        Issuer: C=US, ST=VA, L=Blacksburg, O=ConorCo, CN=conorco.com/emailAddress=conorco@conorco.com
        Validity
            Not Before: Feb  2 22:04:24 2017 GMT
            Not After : Mar  4 22:04:24 2017 GMT
        Subject: C=US, ST=VA, L=Blacksburg, O=U2F Zero, CN=u2fzero.com/emailAddress=conorco@conorco.com
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub: 
                    04:a1:15:5e:63:72:c3:45:ec:97:c1:2a:c2:ba:84:
                    f8:63:f1:ae:6f:f1:7d:80:04:02:a1:9f:86:06:e4:
                    22:a7:c0:2b:e4:e3:44:61:15:3e:fb:d4:13:24:ec:
                    84:b2:54:ec:58:52:91:0d:55:b1:9b:30:79:f5:88:
                    db:f3:05:83:9a
                ASN1 OID: prime256v1
    Signature Algorithm: ecdsa-with-SHA256
         30:44:02:20:20:5c:9e:1d:0d:57:cc:2f:0b:25:f0:97:1c:bf:
         39:8e:a3:b2:0f:9c:97:71:fb:fe:88:f4:f9:a1:03:13:4c:b6:
         02:20:26:ee:da:ae:d8:bf:7c:93:70:0f:cd:1a:26:4c:52:fa:
         e3:86:29:33:50:b1:7a:21:11:2c:03:39:8d:fd:e2:56
-----BEGIN CERTIFICATE-----
MIIB3jCCAYUCAQAwCgYIKoZIzj0EAwIwezELMAkGA1UEBhMCVVMxCzAJBgNVBAgM
AlZBMRMwEQYDVQQHDApCbGFja3NidXJnMRAwDgYDVQQKDAdDb25vckNvMRQwEgYD
VQQDDAtjb25vcmNvLmNvbTEiMCAGCSqGSIb3DQEJARYTY29ub3Jjb0Bjb25vcmNv
LmNvbTAeFw0xNzAyMDIyMjA0MjRaFw0xNzAzMDQyMjA0MjRaMHwxCzAJBgNVBAYT
AlVTMQswCQYDVQQIDAJWQTETMBEGA1UEBwwKQmxhY2tzYnVyZzERMA8GA1UECgwI
VTJGIFplcm8xFDASBgNVBAMMC3UyZnplcm8uY29tMSIwIAYJKoZIhvcNAQkBFhNj
b25vcmNvQGNvbm9yY28uY29tMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEoRVe
Y3LDReyXwSrCuoT4Y/Gub/F9gAQCoZ+GBuQip8Ar5ONEYRU++9QTJOyEslTsWFKR
DVWxmzB59Yjb8wWDmjAKBggqhkjOPQQDAgNHADBEAiAgXJ4dDVfMLwsl8JccvzmO
o7IPnJdx+/6I9PmhAxNMtgIgJu7arti/fJNwD80aJkxS+uOGKTNQsXohESwDOY39
4lY=
-----END CERTIFICATE-----
```

conorpp, 2018-01-06:
This is an older version key.  I recalled my whole U.S. inventory and shipped an updated shipment but I guess Amazon may have failed to ship back some.  Email me your address and order-ID and I'll ship you another token.  Sorry for the mix-up. 

If anyone else happens to get an old token from U.S. market, free feel to send me an email and I can work on getting you a new one.

ryanpcmcquen, 2018-01-06:
How do you get the technical information from a bash shell?

conorpp, 2018-01-06:
I'm not currently aware of a CLI tool that will print out U2F parameters.  One could be fashioned relatively easily using one of these libraries though:

https://github.com/Yubico/libu2f-host
https://github.com/Yubico/python-u2flib-host

ibotty, 2018-01-06:
I assume the other key is also from the old batch, at least it has the same characteristics. I will register it with yubico's testpage as well to be sure though. I will get in touch by mail. Thank you for your kindness.

ellis2323, 2018-03-27:
Hello,

I'm from france and have received an amazon u2f zero. Is this a key from the old batch ?

Registration Data
origin: https://demo.yubico.com
version: U2F_V2
challenge: IbJPqlu671ZlbJHb5ih4JaacwYAsXuiIOB-hOXRaA0A
appId: https://demo.yubico.com

Response Data
clientData: {"typ":"navigator.id.finishEnrollment","challenge":"IbJPqlu671ZlbJHb5ih4JaacwYAsXuiIOB-hOXRaA0A","origin":"https://demo.yubico.com","cid_pubkey":"unused"}
registrationData: 050485fd09b59022e4add20027b01b0e184932184ef445709095c3b3893063944bd8ec4711702e6afe182ee7cd91f99746cd99fed2bda5971bce3ec85ec957f568d424bb07eae738da65e361e1988ede54f5f4b2efc95c64511915c2008ddc771036cbf7948920308201de30820185020100300a06082a8648ce3d040302307b310b3009060355040613025553310b300906035504080c0256413113301106035504070c0a426c61636b73627572673110300e060355040a0c07436f6e6f72436f3114301206035504030c0b636f6e6f72636f2e636f6d3122302006092a864886f70d0109011613636f6e6f72636f40636f6e6f72636f2e636f6d301e170d3137303230323232303432345a170d3137303330343232303432345a307c310b3009060355040613025553310b300906035504080c0256413113301106035504070c0a426c61636b73627572673111300f060355040a0c08553246205a65726f3114301206035504030c0b7532667a65726f2e636f6d3122302006092a864886f70d0109011613636f6e6f72636f40636f6e6f72636f2e636f6d3059301306072a8648ce3d020106082a8648ce3d03010703420004a1155e6372c345ec97c12ac2ba84f863f1ae6ff17d800402a19f8606e422a7c02be4e34461153efbd41324ec84b254ec5852910d55b19b3079f588dbf305839a300a06082a8648ce3d04030203470030440220205c9e1d0d57cc2f0b25f0971cbf398ea3b20f9c9771fbfe88f4f9a103134cb6022026eedaaed8bf7c93700fcd1a264c52fae386293350b17a21112c03398dfde256304402204e7febc4e734f572468673a06c3b61b1783f959964263188a60a4b9d2a6e4410022052484d1c9770b295b3ef9d215ece0b143f2cf6e4c6b5abdda4051d957f29e56f

Attestation Certificate
Certificate:
    Data:
        Version: 1 (0x0)
        Serial Number: 0 (0x0)
    Signature Algorithm: ecdsa-with-SHA256
        Issuer: C=US, ST=VA, L=Blacksburg, O=ConorCo, CN=conorco.com/emailAddress=conorco@conorco.com
        Validity
            Not Before: Feb  2 22:04:24 2017 GMT
            Not After : Mar  4 22:04:24 2017 GMT
        Subject: C=US, ST=VA, L=Blacksburg, O=U2F Zero, CN=u2fzero.com/emailAddress=conorco@conorco.com
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub: 
                    04:a1:15:5e:63:72:c3:45:ec:97:c1:2a:c2:ba:84:
                    f8:63:f1:ae:6f:f1:7d:80:04:02:a1:9f:86:06:e4:
                    22:a7:c0:2b:e4:e3:44:61:15:3e:fb:d4:13:24:ec:
                    84:b2:54:ec:58:52:91:0d:55:b1:9b:30:79:f5:88:
                    db:f3:05:83:9a
                ASN1 OID: prime256v1
    Signature Algorithm: ecdsa-with-SHA256
         30:44:02:20:20:5c:9e:1d:0d:57:cc:2f:0b:25:f0:97:1c:bf:
         39:8e:a3:b2:0f:9c:97:71:fb:fe:88:f4:f9:a1:03:13:4c:b6:
         02:20:26:ee:da:ae:d8:bf:7c:93:70:0f:cd:1a:26:4c:52:fa:
         e3:86:29:33:50:b1:7a:21:11:2c:03:39:8d:fd:e2:56
-----BEGIN CERTIFICATE-----
MIIB3jCCAYUCAQAwCgYIKoZIzj0EAwIwezELMAkGA1UEBhMCVVMxCzAJBgNVBAgM
AlZBMRMwEQYDVQQHDApCbGFja3NidXJnMRAwDgYDVQQKDAdDb25vckNvMRQwEgYD
VQQDDAtjb25vcmNvLmNvbTEiMCAGCSqGSIb3DQEJARYTY29ub3Jjb0Bjb25vcmNv
LmNvbTAeFw0xNzAyMDIyMjA0MjRaFw0xNzAzMDQyMjA0MjRaMHwxCzAJBgNVBAYT
AlVTMQswCQYDVQQIDAJWQTETMBEGA1UEBwwKQmxhY2tzYnVyZzERMA8GA1UECgwI
VTJGIFplcm8xFDASBgNVBAMMC3UyZnplcm8uY29tMSIwIAYJKoZIhvcNAQkBFhNj
b25vcmNvQGNvbm9yY28uY29tMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEoRVe
Y3LDReyXwSrCuoT4Y/Gub/F9gAQCoZ+GBuQip8Ar5ONEYRU++9QTJOyEslTsWFKR
DVWxmzB59Yjb8wWDmjAKBggqhkjOPQQDAgNHADBEAiAgXJ4dDVfMLwsl8JccvzmO
o7IPnJdx+/6I9PmhAxNMtgIgJu7arti/fJNwD80aJkxS+uOGKTNQsXohESwDOY39
4lY=
-----END CERTIFICATE-----

conorpp, 2018-03-28:
Yes keys in Europe Amazon marketplace are affected :(