conorpp/u2f-zero #69: Cannot get U2F Zero to work, no matter the OS or Browser used.

OkaUsagi, 2018-01-04:
I have tried many browsers, and windows as well as linux, with no color change on the LED. It just fades green, with no other color. Pressing the button does give me the blue color, but nothing happens on the U2F side. Any help on what may be wrong? Windows and linux both see it as a U2F Zero, so clearly it's partly working.

ad-m, 2018-01-04:
@OkaUsagi , Wireshark support USB sniffing. Do you can record & publish here session of broken U2F?

OkaUsagi, 2018-01-04:
Yes, i can.

[u2fzero.zip](https://github.com/conorpp/u2f-zero/files/1602282/u2fzero.zip)

conorpp, 2018-01-04:
Have you tried with more than one U2F Zero?

OkaUsagi, 2018-01-04:
Nope, I only ordered one.

OkaUsagi, 2018-01-04:
In addition, I can use the wink command via sudo on my Linux box and that works.

OkaUsagi, 2018-01-04:
But yeah, no luck in figuring out what's wrong on my end,

conorpp, 2018-01-06:
It's probably faulty.  Email me your address and I'll ship another to you.

closed 2018-01-06 by conorpp