conorpp/u2f-zero #71: U2F Zero fails Google's u2f reference code tests

vojnovski, 2018-01-12:
My U2F Zero (with the latest firmware: 21c4f0cb) fails the google u2f tests: https://github.com/google/u2f-ref-code/tree/master/u2f-tests/HID. My yubikey u2f passes almost all of them.

HID Tests U2F Zero:
```
[vv:~/dev/u2f] master ± ./HIDTest 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/U2F Zero@14200000/U2F Zero@0/IOUSBHostHIDDevice@14200000,0'
PASS(test_Idle())
PASS(test_Init())
PASS(test_BasicInit())
PASS(test_Unknown(U2FHID_SYNC))
PASS(test_InitOnNonBroadcastEchoesCID())
PASS(test_InitUnderLock())
PASS(test_InitSelfAborts())
.PASS(test_InitOther())
PASS(test_OptionalWink())
PASS(test_Lock())
PASS(test_Echo())
CHECK_LE fail at test_LongEcho[148]:sent > .075: zsh: abort      ./HIDTest
```
HID Tests Yubikey:
```
[vv:~/dev/u2f] master ± ./HIDTest 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS01@14100000/Security Key by Yubico@14100000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14100000,0'
PASS(test_Idle())
PASS(test_Init())
PASS(test_BasicInit())
PASS(test_Unknown(U2FHID_SYNC))
PASS(test_InitOnNonBroadcastEchoesCID())
PASS(test_InitUnderLock())
PASS(test_InitSelfAborts())
PASS(test_InitOther())
PASS(test_OptionalWink())
PASS(test_Lock())
PASS(test_Echo())
PASS(test_LongEcho())
PASS(test_Timeout())
PASS(test_WrongSeq())
PASS(test_NotCont())
PASS(test_NotFirst())
PASS(test_Limits())
PASS(test_Busy())
PASS(test_Interleave())
PASS(test_LeadingZero())
PASS(test_Idle(2.0))
PASS(test_NothingOnChannel0())
PASS(test_OnlyInitOnBroadcast())
PASS(test_Descriptor())
```

U2F Tests U2F Zero:
```
[vv:~/dev/u2f] master ± ./U2FTest 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/U2F Zero@14200000/U2F Zero@0/IOUSBHostHIDDevice@14200000,0'
PASS(check_Compilation())
PASS(test_Version())
PASS(test_UnknownINS())
PASS(test_WrongLength_U2F_VERSION())
PASS(test_WrongLength_U2F_REGISTER())
PASS(test_BadCLA())
PASS(test_Enroll(0x6985))

Touch device and hit enter..

CHECK_EQ fail at test_Enroll[134]:expectedSW12 != U2Fob_apdu(device, 0, U2F_INS_REGISTER, U2F_AUTH_ENFORCE, 0, string(reinterpret_cast<char*>(&regReq), sizeof(regReq)), &rsp): zsh: abort      ./U2FTest
```

U2F Tests Yubikey:
```
[vv:~/dev/u2f] master ± ./U2FTest 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS01@14100000/Security Key by Yubico@14100000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14100000,0'
PASS(check_Compilation())
PASS(test_Version())
PASS(test_UnknownINS())
PASS(test_WrongLength_U2F_VERSION())
PASS(test_WrongLength_U2F_REGISTER())
PASS(test_BadCLA())
PASS(test_Enroll(0x6985))

Touch device and hit enter..

PASS(test_Enroll(0x9000))
PASS(test_Sign(0x6985))
PASS(test_Sign(0x6985, true))
PASS(test_Sign(0x6a80))
PASS(test_Sign(0x6a80))

Touch device and hit enter..

PASS(test_Sign(0x6985, true))
PASS(ctr1 = test_Sign(0x9000))
PASS(test_Sign(0x6985))

Touch device and hit enter..

PASS(ctr2 = test_Sign(0x9000))
CHECK_EQ fail at test_Sign[235]:expectedSW12 != real: zsh: abort      ./U2FTest
```

conorpp, 2018-01-12:
Can you register your token [here](https://demo.yubico.com/u2f) and post the technical information?

ygator, 2018-01-12:
HIDTest
Mine fails as well, but if you run with -a it passes most tests but:
CHECK_LE fail at test_LongEcho[148]:sent > .075: (continuing -a)
CHECK_GE fail at test_LongEcho[149]:received < .020: (continuing -a)

Looking into the tests it says
  // Expected transfer times for 2ms bInterval.
  // We do not want fobs to be too slow or too aggressive.
and it wants the time to be >=.020 and <=.075
My device: sent: 0.0966043, received: 0.0190652
So the send took longer than .075 and the receive was faster than .020

U2FTest
It fails a lot.  I did not look into it much, but the program does core dump when it fails.

conorpp, 2018-01-13:
With latest firmware, U2FTest should pass.  The LongEcho tests will fail since the MCU isn't fast enough

vojnovski, 2018-01-16:
@conorpp, here's the technical data from https://demo.yubico.com/u2f?tab=register: 

```
Login Data
username: tralalaxvv
password: tralalaxvv

Registration Data
origin: https://demo.yubico.com
version: U2F_V2
challenge: oSDLDo3jhUFqc-4NA0mq8yeqs30cpnjKY1YrljexyPo
appId: https://demo.yubico.com

Response Data
clientData: {"typ":"navigator.id.finishEnrollment","challenge":"oSDLDo3jhUFqc-4NA0mq8yeqs30cpnjKY1YrljexyPo","origin":"https://demo.yubico.com","cid_pubkey":"unused"}
registrationData: 050442302a0061eb7680e8898fc557ba3c655d83e5c9071fced0e9550212793d5c6b7b99f8eee1ee5258012d9ad8ac4bbafb01d028446c403e26e0e3597135b019c12c9d941e8aa5f495e27fac434d6e19caf2ed9f217aadceeef40e6d35842c62730cc45227847861e9401aa84b8a308201a330820148020900b497b1d924280d56300a06082a8648ce3d0403023058310b3009060355040613024d4b310f300d06035504080c06536b6f706a65310f300d06035504070c06536b6f706a6531153013060355040a0c0c4861636b6c6162204b494b413110300e06035504030c076861636b6c61623020170d3138303130353231323630335a180f32303637313232343231323630335a3058310b3009060355040613024d4b310f300d06035504080c06536b6f706a65310f300d06035504070c06536b6f706a6531153013060355040a0c0c4861636b6c6162204b494b413110300e06035504030c076861636b6c61623059301306072a8648ce3d020106082a8648ce3d03010703420004fed5b5d61d4e94e375fe597452596ce582df9fc892736378e744dc0bf94d50c2ac40b9ceef991ce715279e2b773e53109562f5a2714620844d00914b73c8e416300a06082a8648ce3d0403020349003046022100d6a3522968da13d69be3f1e707a36cac5d606c478c96872dfb3d8d7a84d5a4e1022100848e45519d1edb12d5432ab3d41cb8aa9ebf58fd8f929a421e9778445cd5615f30460221009c298c7d1b16ff812ac20f45a40b852f882587daac385106b4e8f04afeed67f60221008a2cb050ac5c364dfef86463255a6016f365e7f3e53b7a70278c29c65806f8c7

Attestation Certificate
Certificate:
    Data:
        Version: 1 (0x0)
        Serial Number: 13013065194482961750 (0xb497b1d924280d56)
    Signature Algorithm: ecdsa-with-SHA256
        Issuer: C=MK, ST=Skopje, L=Skopje, O=Hacklab KIKA, CN=hacklab
        Validity
            Not Before: Jan  5 21:26:03 2018 GMT
            Not After : Dec 24 21:26:03 2067 GMT
        Subject: C=MK, ST=Skopje, L=Skopje, O=Hacklab KIKA, CN=hacklab
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub: 
                    04:fe:d5:b5:d6:1d:4e:94:e3:75:fe:59:74:52:59:
                    6c:e5:82:df:9f:c8:92:73:63:78:e7:44:dc:0b:f9:
                    4d:50:c2:ac:40:b9:ce:ef:99:1c:e7:15:27:9e:2b:
                    77:3e:53:10:95:62:f5:a2:71:46:20:84:4d:00:91:
                    4b:73:c8:e4:16
                ASN1 OID: prime256v1
    Signature Algorithm: ecdsa-with-SHA256
         30:46:02:21:00:d6:a3:52:29:68:da:13:d6:9b:e3:f1:e7:07:
         a3:6c:ac:5d:60:6c:47:8c:96:87:2d:fb:3d:8d:7a:84:d5:a4:
         e1:02:21:00:84:8e:45:51:9d:1e:db:12:d5:43:2a:b3:d4:1c:
         b8:aa:9e:bf:58:fd:8f:92:9a:42:1e:97:78:44:5c:d5:61:5f
-----BEGIN CERTIFICATE-----
MIIBozCCAUgCCQC0l7HZJCgNVjAKBggqhkjOPQQDAjBYMQswCQYDVQQGEwJNSzEP
MA0GA1UECAwGU2tvcGplMQ8wDQYDVQQHDAZTa29wamUxFTATBgNVBAoMDEhhY2ts
YWIgS0lLQTEQMA4GA1UEAwwHaGFja2xhYjAgFw0xODAxMDUyMTI2MDNaGA8yMDY3
MTIyNDIxMjYwM1owWDELMAkGA1UEBhMCTUsxDzANBgNVBAgMBlNrb3BqZTEPMA0G
A1UEBwwGU2tvcGplMRUwEwYDVQQKDAxIYWNrbGFiIEtJS0ExEDAOBgNVBAMMB2hh
Y2tsYWIwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAAT+1bXWHU6U43X+WXRSWWzl
gt+fyJJzY3jnRNwL+U1QwqxAuc7vmRznFSeeK3c+UxCVYvWicUYghE0AkUtzyOQW
MAoGCCqGSM49BAMCA0kAMEYCIQDWo1IpaNoT1pvj8ecHo2ysXWBsR4yWhy37PY16
hNWk4QIhAISORVGdHtsS1UMqs9QcuKqev1j9j5KaQh6XeERc1WFf
-----END CERTIFICATE-----
```

vojnovski, 2018-01-20:
Some further comments: 

* The token works well with the most prominent u2f services, so real-world usage is not a problem.
* During the U2FTest tests, as evidenced in the first comment, it fails at Enroll. The LED never turns red as to signal need for a user presence check. It certainly might be a timing issue as well.

Jerry280989322, 2019-01-04:
Hi,it seems that I met with the same problem.Have you solve the problem?  
 The HID and U2F Test results are the same as yours. When I try to test it in the [https://demo.yubico.com/u2f?tab=register](url), the LED on the key  turns red , and the result of registration is  "Register Failed!!"