conorpp/u2f-zero #72: Bought two keys. Neither will work on my mac or windows machines.

LordPowerbang, 2018-01-16:
On the yubico tester website it just says "timed out" but it happens the instant I press the button. Have tried 2 keys on two operating systems with USB 2.0 and 3.0. Very confused.

Davidsg33k, 2018-01-16:
Hi LordPowerbang,
When did you order the keys?

It would seem there was an issue with some of them recently.  "I experienced the same issue with the two keys i received.

I bet you can send "conorpp" a message to see what can be done. 
 "perhaps on amazon if you got it there"
"I have to wait for the return funding to show up before i can order replacements."

Hopeful i will have them soon been waiting since dec 28th to test these.

best of luck, seems the bad batch got us both   "here is looking forward to testing the good ones when they come..

best regards 
David Gaspar

conorpp, 2018-01-17:
There was a small shipment of faulty tokens to Amazon in U.S..  I've since recalled it and replaced the inventory.

If you made an order sometime in December or first week of January, there's a chance it's faulty because of my bad shipment.  

If this is the case, please email me your order ID or just send me a message through Amazon and I'll refund you.

Sorry about that!!

LordPowerbang, 2018-01-17:
Thanks, Conorpp. I just sent you a message through Amazon. Mine were indeed purchased on 12/27/17, and received them 1/2/18, so it sounds like I was in the right window. I'm glad you're on top of it! Thanks a bunch!

closed 2018-02-03 by conorpp