conorpp/u2f-zero #74: EFM8 is not locked at the end of the programming/flashing process

vojnovski, 2018-01-21:
Is there a reason why the EFM8 is not locked at the end of the programming process?

It could be done as specified here: https://www.silabs.com/community/mcu/8-bit/knowledge-base.entry.html/2016/01/13/locking_flash_using-RdLy, setting 0xFBFF (the EFM8 Lock Byte: https://www.silabs.com/documents/public/data-sheets/efm8ub1-datasheet.pdf) to 0x00 with the linker.

szszszsz, 2018-07-25:
For completeness: currently `0xFBFF` byte is conditionally set to `-32` on each firmware start, at [src/eeprom.c#L39](https://github.com/conorpp/u2f-zero/blob/6a177cad311c8710dae148fb6e2b7ffc3a8fd3b9/firmware/src/eeprom.c#L39)

https://github.com/conorpp/u2f-zero/blob/6a177cad311c8710dae148fb6e2b7ffc3a8fd3b9/firmware/src/eeprom.c#L31-L40

szszszsz, 2018-08-03:
Just to explain `-32` - according to https://www.silabs.com/documents/public/reference-manuals/EFM8UB1-RM.pdf this locks first 32 pages (32x512=16kB) of user flash.

szszszsz, 2018-08-03:
@vojnovski Do you know, whether `0` would work the same and lock all the pages?

vojnovski, 2018-10-30:
Hi @szszszsz, yes, from the same manual, setting it to `0` locks all user flash pages.