conorpp/u2f-zero #78: Firefox Quantum unable to authenticate with U2F-zero

GitHubMad, 2018-03-22:
Hello
On Windows 7 or 10, my U2F-zero works well with Chrome or FF 52 ESR + [U2F Support Add-on](https://addons.mozilla.org/fr/firefox/addon/u2f-support-add-on/) : Great !

But trying on FF 57 to 59 :
- **Registering works** on most of the [U2F sites](http://www.dongleauth.info/) I tested (often need to tweak UA to Chrome)
- But **Authent for login always fails** (either with UA Chrome or not) : I see FF sending many trial requests to the Key but none succeed

On my Firefox Quantum I activated **security.webauth.u2f** and I don't have any U2F Plug-In (as U2F support is now native and "U2F Support Add-on" is not available for Quantum).

Firefox Quantum is said to support only [FIDO U2F devices using Web Authentication APIs (as part of FIDO 2) versus FIDO U2F APIs](https://www.yubico.com/2017/11/how-to-navigate-fido-u2f-in-firefox-quantum/).

Does it mean U2F-zero is not compatible with FIDO 2 (and so, with Firefox) ?
Any hope it will be ?

I bought mine [here](https://www.amazon.fr/U2F-Zero-U2-F/dp/B01L9DUPK6) around Xmas.

NOTE: even FF ESR will soon run into the same trouble, as it will jump to FF 60 when available (in a few monthes).

galadran, 2018-05-10:
I've just upgraded to FF60 and I've run into exactly this issue. I can register my device correctly, both on the Firefox test [page](https://webauthn.bin.coffee/) and on Github, however on both pages I can't login. 

Firefox sees the request and prompts me to press a button on the token, however pressing the button doesn't result in anything happening. The LED doesn't go orange like it does for a registration request, although sometimes it starts blinking green. 

On Chrome, the U2FZero works perfectly.

Is there any fix for this this? I'd really like use my U2FZero with Firefox.

conorpp, 2018-05-10:
In the FIDO 2 spec, the transport protocols (e.g. USB, NFC) are pretty much
the same so U2F still works alongside FIDO 2.  That is, if U2F is
implemented.  I believe FF60 does not implement U2F despite it having FIDO
2.  That’s odd that the register works but not authenticate..

I’m working on a FIDO 2 implementation now and will hopefully have it out
soon.  Old U2F zero devices won’t work since they won’t have enough memory
but a pin compatible upgrade to the efm8ub3 chip might work out.

nextgenthemes, 2018-05-11:
@conorpp 

> I believe FF60 does not implement U2F despite it having FIDO

There are several switches in the about:config that suggest otherwise. There is a switch for UF2 and a softoken switch that one demo tells you to switch on. I bought the U2FZero to work with Firefox and was under the impression that it will work. I did not get what was really new in 60 when the other then they enabled a switch to be turned on by default.

All I can say is I trail and errored my was through all this links on this page https://wiki.mozilla.org/Security/CryptoEngineering and actually had the blinking working and for the https://webauthn.io/ demo the registration actually works and logs me in at the same time but then the login fails.

I also have found something somewhere to enabled the some USB think on Ubuntu Linux and i am not sure if its even needed. I have found **ABSOLUTELY NOTHING** on the official homepage about anything. I am pretty disappointed with this because its a product that is being sold with no goto place what the actual state of things is what works and how to get it to work ... nothing. But I guess its me again being cheap. I love the open hardware and stuff but there should be some info out there.

I have just successfully tested the U2F demo in chromium on ubuntu 18.04. But this webauthn demo https://webauthn.bin.coffee/ fails with this 

> NotSupportedError: The user agent does not implement a password store.

Not really feel right to do hours of research and trail error end up on Github.

I thought:

> Open Standard - Open Hardware - Webauthn is basically U2F 2.0 or something like that and read devices that support U2F will automatically work with webauthn

But I guess not. How about telling this to people before they purchase this device? How about putting any info up on the site?

I thought I buy something that just works. And how to make it work later? Firmware upgrade is possible? Did not found anything about that. Instead searching for this leads me to the wiki to build my own and "program" it. I am not sure what if this mixes building and flashing of a bought one or whatever its just confusing.

> Old U2F zero devices won’t work since they won’t have enough memory but a pin compatible upgrade to the efm8ub3 chip might work out.

Does that mean my device I just bought might be old and will just not work or i will need to purchase some chip and put in on there?

conorpp, 2018-05-11:
I've tested on FF60 on https://u2f.bin.coffee/ and Github (register and authenticate) with latest U2F firmware and it seems to work fine.

@galadran @nextgenthemes when did you acquire U2F Zero?

nextgenthemes, 2018-05-11:
@conorpp 

> Amazon page

How about the homepage? I do not use Amazon and I bought it there.

>  I agree with you

OK great how about actually addressing my comment and replying to my questions? Most importantly: Will there be a firmware upgrade that I can apply that will fix this? 

I will mail you for a refund for sure, this is not acceptable to me. With a heads up for sure but not like this. Should have bought some mainstream device and plug it on and be happy.

> when did you acquire U2F Zero?

Very recently. I just tested the demo you linked again and the reg work then for auth it blinks but pushing it will do nothing at all.

conorpp, 2018-05-11:
The European stock is out of date and suffers from a couple bugs discovered around the start of this year.  The current firmware (and U.S. sold devices) should not have any issues with Firefox.

I'm working to update EU stock soon.  I have a notice on the Amazon pages but nothing on https://u2fzero.ch/.  Sorry!

nextgenthemes, 2018-05-16:
And again not addressing anything I asked about. And there is another utterly crazy crazy email exchange going on with me and this @conorpp dude who seem to unable to comprehend anything and after I told him I bought a single device he insisted that I am a reseller and wants so replace some imaginary stock I have. Seriously unfit! Anyone reading this do not buy this shit device from this amateurish dude who cant even read emails. Not sure why this surprises me given the amount of dedication he showed here to actually answer things and that the website has zero info on anything ...

rickard-von-essen, 2018-05-16:
Wow! That was rude.

MaPePeR, 2018-05-16:
> There are several switches in the about:config that suggest otherwise.

These are switches for **experimental** features and are not guaranteed to work. :man_shrugging: 
And Firefox not working correctly is not a U2Fzero-only problem anyway. It does not allow authentication with google with ANY U2F-device - regardless of who made/sold it, for example. 
Firefox did not even try to support U2F until very very recently, so a device that is a lot older than the firefox support can not really be expected to work flawlessly with firefox, anyway.

Also you cannot really blame anyone besides yourself for buying something without doing research about it before. :man_shrugging: 

> I have just successfully tested the U2F demo in chromium on Ubuntu 18.04. 

So... you bought a product that works.  

I'm pretty sure everyone here would be willing to try figure out, what your problem is and how to fix it, but the tone of your messages makes taking you seriously and communicating with you very very hard (at least for me).

*Disclaimer: Not associated with u2fzero/conorco/etc.*

nextgenthemes, 2018-05-16:
Yeah that how I roll. When there is a reason for it I tell the facts but that so "rude". I already got a down-vote on my very extensive https://github.com/conorpp/u2f-zero/issues/78#issuecomment-388244301 for pointing out facts. Got all just ignored by @conorpp but seems to be a trend online to just ignore and skip over things ... but not dare anyone demand anything form a product that is sold for money. No no be all nice a cuddly with people who show you nothing for what you could possible respect them for.

I am not saying I could do this better. That's why I am not doing it! I could compile a simple basic straight forward list of information I would expect to see on the website I am rally not feeling like helping out.

And just so nobody get this wrong its not about that there is a faulty batch out, that can happen. Its about *everything around it*. The fact that you can not update the firmware via USB, that you are not told this b4 you buy it, that I have not been taken seriously here from the start, despite him offering me a refund and a replacement I do not feel he is paying attention at all. Its looks like a amateurish alpha operation to me and I can not recommend anyone to buy it of the creator does not show more dedication but at the same time sells it as a ready product.

And that email exchange, its so crazy I still can't believe this ... but its all just me. I am so "rude".

@MaPePer

>> I have just successfully tested the U2F demo in chromium on Ubuntu 18.04.

> So... you bought a product that works.

So because It works in a browser thank you do not use I should be happy with it. Are you kidding me? Not even he thinks that he already told that there is a faulty batch out!

> I'm pretty sure everyone here would be willing to try figure out,

No one needs to figure anything out. I need a Arduino to just upgrade the firmware or just a new device that's it.

MaPePeR, 2018-05-16:
There is nothing rude about just telling facts. But that is not what you do.

> [...] but not dare anyone demand anything form a product that is sold for money.  No no be all nice a cuddly with people who show you nothing for what you could possible respect them for.

Being respectful to oneanother is one of the core principals of communicating between adults. You seem to disagree on that and require "proof" that someone is to be respected by you.

> The fact that you can not update the firmware via USB 

(I think) it would be horrible insecure if that would be possible.

> that you are not told this b4 you buy it

because (at least to me) it seems obvious, that this "feature" would break the product. Also there is no place where it says, that you CAN do that, so you cannot really expect a feature that was never even advertised?

>  that I have not been taken seriously here from the start

You just wrote that you don't take anyone seriously and refuse to act like a decent human being, before they show you something "you could possible respect them for", but demand to be taken seriously... I hope you can realize how that is a problem.

> So because It works in a browser that I do not use I should be happy with it.

The browser you use did not include the feature you want to use as of very recently.

> its also not working with the newest and coming standard 

You cannot expect something you buy, that is already a little bit older, to work flawlessly with the newest coming standard for which it was not made.

Buy a car today and complain that its not autonomous when autonomous cars become standard.

conorpp, 2018-05-16:
Thank you all for your comments but it's getting off topic so I'm going to lock and close this thread.  Feel free to open other issues that are related to the open source project.

For sales/refunds/product complaints, please use the new support page.  FAQ will be updated with common issues that come up.

https://u2fzero.com/#support

closed 2018-05-16 by conorpp