conorpp/u2f-zero #82: unable to register

e3bo, 2018-06-10:
This could be user error, but I'm not able to register either of the 2 u2fzeros I just received last week from Amazon on either GitHub, Google, or https://u2f.bin.coffee/. On every registration attempt, the light never turns orange, although it goes from green to blue when I hold the button down. I've tried with Firefox 60 and Chromium 67 on Arch Linux, using all 3 USB ports on my laptop. It seems to me I have a similar problem to #69, which was attributed to faulty devices. However, I was wondering if someone could explain to me how I could confirm that, or suggest what I might be doing wrong.

conorpp, 2018-06-10:
When did you order them?  In the U.S.?

e3bo, 2018-06-10:
I ordered them on June 3 in the U.S.  They shipped on June 6. The return address on the shipping envelope is Lexington, KY.

I'm going to try using them in a Windows computer in a few minutes.

MaPePeR, 2018-06-10:
Just a very random guess: did you add ehe udev rule that is required to operate u2f devices on linux? (Not sure if they are added by default)

e3bo, 2018-06-10:
That was it. Thanks for your patient help. I looked at the wiki last night but somehow missed that. For anyone else with this problem, see instructions on this page:

https://github.com/conorpp/u2f-zero/wiki/Using-U2F-Token-on-Linux-(udev)

closed 2018-06-10 by e3bo