conorpp/u2f-zero #86: Flashing of retail version

char32, 2018-07-28:
I wanted to flash a current version on a u2fkey.

What I tried is to bring a bought version of the u2fzero into a bootloader mode. But it does not seem to work.
Is it disabled?

Is there a manual on how to connect a programmer (like the SLSTK2000a or similar) to the key without soldering wires to it? I did not find anything in the wiki.

Thanks a lot! And keep up the great work :-)

char32, 2018-07-28:
Thanks for the fast answer! So the booloader is removed due to space issues. Makes sense.
But it should be possible to use a programmer, right? I just need to get the connection right.

Thanks again!

ad-m, 2018-07-28:
See https://github.com/conorpp/u2f-zero/wiki/Building-a-U2F-Token#using-a-programmer .

char32, 2018-07-28:
Thanks! I'll try to build something out of header connectors or so to get a stable connection between programmer and stick without soldering wires to the stick.

ad-m, 2018-07-28:
I'm not an electronics. Can you use some light glue / plasticine for that?

char32, 2018-07-29:
Yeah I am more on the software development side too :-) Good idea with the plasticine! I'll try that.
Something like https://www.segger.com/products/debug-probes/j-link/accessories/adapters/10-pin-needle-adapter/ would be perfect, but they are expensive.

rickard-von-essen, 2018-07-29:
I think I cut one of [these](https://eu.mouser.com/ProductDetail/798-A110PA254DSA71) and soldered together a cable.