conorpp/u2f-zero #91: Firefox Quantum unable to authenticate with U2F-zero

oupala, 2018-10-04:
This issue is rebirth of issue #78 as I'm having the same problem.

I'm using Ubuntu 18.04, the last stable. I'm also using Firefox 62.0.3. My u2f key, a new one supposedly without the bug that affected previous u2fzero keys.

When I use the [yubico test site](https://demo.yubico.com/u2f) I can't register, neither can I login. It seems to be no interaction between the u2f key and Firefox.

When I use Chrome 69, I have no problem registering and logging in.

Could you please help me troubleshooting this problem?

oupala, 2018-10-04:
I just read issue #78 and I went to the test site mentionned bu @conorpp: [u2f.bin.coffee](https://u2f.bin.coffee/).

I saw that I had to enable two properties in Firefox:

To use this to test U2F support in Firefox using its built-in software token, do the following:

> - Open the Firefox advanced preferences in about:config.
> - Search for "webauth"
> - Enable (value=True) these two:
>   - `security.webauth.u2f`
>   - `security.webauth.u2f_enable_softtoken`

With theses properties enabled, everything works fine on [u2f.bin.coffee](https://u2f.bin.coffee/) website, and on [yubico test site](https://demo.yubico.com/u2f).

Could you please explain us why are these properties disabled by default on Firefox for linux? A friend of mine tested it on windows and there was no need to change properties in Firefox for windows...

YakBizzarro, 2018-10-04:
Mozilla implemented U2F API in Firefox, but they never enabled that by default, since it has been obsoleted by WebAuthn, which is supported by default

oupala, 2018-10-05:
In fact, in my case, enabling `security.webauth.u2f_enable_softtoken` wasn't necessary. I only had to enable `security.webauth.u2f`.

I also had to enable this u2fzero key in udev as mentioned in the [Using U2F Token on Linux (udev)](https://github.com/conorpp/u2f-zero/wiki/Using-U2F-Token-on-Linux-(udev)) wiki page.

So using u2fzero does not work out of the box with linux, but it is quite easy to make it work in 2 steps.

oupala, 2018-10-05:
@YakBizzarro Does the WebAuthn include the u2f keys and protocol? Is there a link between WebAuthn and u2f?

YakBizzarro, 2018-10-05:
It's well explained here: https://www.imperialviolet.org/2018/03/27/webauthn.html

oupala, 2018-10-09:
Thanks @YakBizzarro for the link. It is interesting.

closed 2018-10-09 by oupala