conorpp/u2f-zero #96: Unable to register with any U2F device.

Jerry280989322, 2019-01-03:
I've been tring to make the U2F-zero interact with U2F-host(https://github.com/Yubico/python-u2flib-host),but it doesn't work any more. The result is as following:

~$ u2f-register -i reg.json http://localhost:8081

Touch the U2F device you wish to register...

Unable to register with any U2F device.

Is it because the slots is full?? or else? It troubles me a lot.I would be very thankful if someone could help me out.

conorpp, 2019-01-03:
what operating system and browser are you using?

Jerry280989322, 2019-01-03:
Ubuntu 18.04.1 LTS  
No browser, just the command line.

conorpp, 2019-01-03:
did you set a udev rule?
https://github.com/conorpp/u2f-zero/wiki/Using-U2F-Token-on-Linux-(udev)

Jerry280989322, 2019-01-03:
> did you set a udev rule?
> https://github.com/conorpp/u2f-zero/wiki/Using-U2F-Token-on-Linux-(udev)

Oh god, I haven't do that.Thanks~

Jerry280989322, 2019-01-03:
Oh....I meet with another problem.There is no    70-u2f.rule    in my /etc/udev/rules.d, but a  70-snap.core.rules.
Did I miss someting?

conorpp, 2019-01-03:
You have to create the file

Jerry280989322, 2019-01-03:
OK~

Jerry280989322, 2019-01-03:
It still doesn't work... Besides, when I do the HID-test, it meet with a problem  as follows:
PASS(test_Idle())
PASS(test_Init())
PASS(test_BasicInit())
PASS(test_Unknown(U2FHID_SYNC))
PASS(test_InitOnNonBroadcastEchoesCID())
PASS(test_InitUnderLock())
PASS(test_InitSelfAborts())
PASS(test_InitOther())
PASS(test_OptionalWink())
PASS(test_Lock())
PASS(test_Echo())
CHECK_LE fail at test_LongEcho[148]:sent > .075: 已放弃

conorpp, 2019-01-03:
yes it's a little slow, that particular test shouldn't be an issue.  What about the U2F test?

Jerry280989322, 2019-01-03:
> yes it's a little slow, that particular test shouldn't be an issue. What about the U2F test?

Oh,I haven't try it yet. I will try it after lunch.  :)

Jerry280989322, 2019-01-03:
Here is the result of U2FTest, it seems that there is some problems with enroll:

$ sudo ./U2FTest /dev/hidraw6
PASS(check_Compilation())
PASS(test_Version())
PASS(test_UnknownINS())
PASS(test_WrongLength_U2F_VERSION())
PASS(test_WrongLength_U2F_REGISTER())
PASS(test_BadCLA())
PASS(test_Enroll(0x6985))

Touch device and hit enter..

CHECK_EQ fail at test_Enroll[134]:expectedSW12 != U2Fob_apdu(device, 0, U2F_INS_REGISTER, U2F_AUTH_ENFORCE, 0, string(reinterpret_cast<char*>(&regReq), sizeof(regReq)), &rsp): 已放弃

conorpp, 2019-01-04:
Where/when did you get your token?

Jerry280989322, 2019-01-04:
about 2016.11    I guess.  It was given by my friend.

Jerry280989322, 2019-01-04:
when I try to test it in the ,[ https://demo.yubico.com/u2f?tab=register](url)，it returns that 

Registration failed!
Make sure you have a U2F device connected, and try again.

 Traceback (most recent call last):
  File "/root/python-u2flib-server-demo/examples/yubiauth_server.py", line 161, in __call__
    raise Exception("FIDO Client error: %s" % error)
Exception: FIDO Client error: 1 (OTHER ERROR)

 And during the time, the LED on the board turn red from green , what does the red light mean?

conorpp, 2019-01-06:
There was an issue with a particular U2F "CHECK-ONLY" command, which is used to see if a key handle is owned by a U2F token.  Needed to be used when there are multiple U2F tokens, some services always use it first.  This has since been fixed but it's likely your token has this problem :(

Jerry280989322, 2019-01-06:
> There was an issue with a particular U2F "CHECK-ONLY" command, which is used to see if a key handle is owned by a U2F token. Needed to be used when there are multiple U2F tokens, some services always use it first. This has since been fixed but it's likely your token has this problem :(
> 
> You can exchange it with the channel you or your friend got it from.

You mean the old version of firmware  has this problem and it has been fixed in the latest firmware now?

conorpp, 2019-01-06:
yes

Jerry280989322, 2019-01-08:
Can I update the firmware into the u2f-zero by myself?
I don't have a programmer so I have to use the bootloader method.when I use the command "python client.py bootloader"   ,it turns out that 

~/u2f-zero/tools/u2f_zero_client$ python client.py bootloader
opened  None
If this device has an enabled bootloader, the LED should be turned off.

But I found that the LED is still on(green). Am I do the right steps if I want to change the u2f-zero into bootloader mod ? @conorpp