conorpp/u2f-zero #99: U2F Device Failure

Immortalin, 2019-03-26:
My U2F zero stopped working sometimes last week. Would remain on idle only i.e. when browser gives prompt the light remains green. Testing with the test suite returns the following:

```
PASS(test_Idle())
PASS(test_Init())
PASS(test_BasicInit())
PASS(test_Unknown(U2FHID_SYNC))
CHECK_EQ fail at test_InitOnNonBroadcastEchoesCID[469]:0 != U2Fob_receiveHidFrame(device, &r, 1.0): Aborted (core dumped)
```

rlarranaga, 2019-03-28:
I had the same issue. Any chance Chrome Broke Support with the last release? I am on 73.0.3683.86 on windows 10.
VERY annoying.
Thanks!
Regards

conorpp, 2019-03-28:
Unfortunately there's a bug in U2F Zero where it returns an incorrect error code.  This has been fine in the past and has gone unnoticed, until browsers have updated to a more strict checking.  So now U2F Zero won't authenticate properly on most browsers.

This has been fixed on Solo version:
https://github.com/solokeys/solo

U2F Zero can't be updated easily and would recommend getting a refund through the proper sales channel and getting a Solo key :)

rlarranaga, 2019-03-28:
Hey Connor...Not great to hear...i have a bunch of sites on u2f-zero....
I am already a backer of solo, but still waiting for my keys, since i chose solo tap...
How hard is "can't be updated easily"?
Thanks!
Regards

conorpp, 2019-03-28:
I believe it will still work with the Brave browser, so I recommend removing U2F Zero as a factor for now.  If you email me a photo of your U2F Zero, I'll mail you a free Solo USB-A replacement :) 

conorpp94@gmail.com

closed 2019-03-28 by Immortalin